hideakitai/MPU9250 #2: Not Returning Proper Yaw Readings

Jgunde, 2019-01-05:
I am using an MPU9250 with an Arduino Pro Mini. I am testing with two 9250s so it is unlikely both are defective. I have tried to calculate yaw values from the sensor but couldn't get accurate results. I performed all the calibration steps from the calibration example in many different ways but it still hasn't worked. Any ideas what could be wrong?

P.S.  The yaw values aren't correct because when I turn the sensor 90 degrees it reads only turning 70. But when I turn it another 90 degrees it reads turning 110 degrees. 

Side note: The values returned are "precise" because if I rotate the sensor and then rotate it back to the same position the yaw values are the same.

Here is what the calibration returns:

Mag Calibration: Wave device in a figure eight until done!
mag x min/max:
-85
-249
mag y min/max:
221
129
mag z min/max:
358
303
Mag Calibration done!
AK8963 mag biases (mG)
-299.30, 313.64, 568.25
AK8963 mag scale (mG)
0.63, 1.12, 1.91

hideakitai, 2019-03-31:
I'm so sorry for late reply... please calibrate mag value in zero degree to detect gravity value exactly. Thanks!

closed 2019-03-31 by hideakitai

korbinianbauer, 2020-07-12:
Your scale values look pretty excessive, you'd expect values much closer to 1, unless the Mag is very close to a chunk of steel or another source of disturbance (A "normal" range is 0.8-1.2, but not 0.63 or 1.91 )

Make sure your pointing the sensor in every direction. Think of it like standing inside an imaginary sphere and sweeping over every spot with a flashllight (aka the MPU).

You can always extend the calibration duration by increasing the `sample_count` in MPU9250.h

`        // shoot for ~fifteen seconds of mag data
        if      (Mmode == 0x02) sample_count = 128;  // at 8 Hz ODR, new mag data is available every 125 ms
        else if (Mmode == 0x06) sample_count = 1500;  // at 100 Hz ODR, new mag data is available every 10 ms`