hideakitai/MPU9250 #4: Could you share your eeprom library?

enginergun, 2019-03-11:
Firstly thanks for your excellent mpu9250 library. It is very usefull. I want to write sensor's offsets to eeprom but your calibration_eeprom sketch is giving library error, i found a library (https://github.com/thijse/Arduino-EEPROMEx) but it also giving error. " class EEPROMClassEx has no member named 'commit' and has no member named 'begin'. " How can i solve this problem? 

Thank you.

hideakitai, 2019-03-30:
I'm sorry for late reply.
What platform do you use? Some platform may not be able to compile `calibration_eeprom.ino`.

enginergun, 2019-03-30:
Thanks for reply it's not late for me :) I'm using arduino mega 2560

hideakitai, 2019-03-30:
I have added an example for Uno/Mega board. Though I haven't check if it works, please check it out!

closed 2019-03-30 by hideakitai