hideakitai/MPU9250 #6: Wrong magnetic resolution

gmikitiuk, 2019-06-16:
Hi,

Are you sure u got proper calculation for magnetic resolution? Just checked your comments and u clearly state 0.6mG/LSB which is incorrect, it's 0.6uT/LSB(according to datasheet) which gives 6mG/LSB.

Both gyro and acc resolution seems to be fine but magnetic gives wrong results

hideakitai, 2019-06-16:
Hi, the calculation and most comments are just ported from original one. Which comment?

gmikitiuk, 2019-06-16:
enum class MFS { M14BITS, M16BITS }; // 0.6mG, 0.15mG per LSB <--- This one

Just went deeper into the code and it seems that calculation returns correct values of mG, comments is just misleading

akshayacharya97, 2020-04-28:
Hello, please send the calibration values of the mag bias and the mag scale that you got. Ours seems to be changing with each calibration.

hideakitai, 2020-11-04:
Thanks! I've removed confusing comment.

closed 2020-11-04 by hideakitai