hideakitai/MPU9250 #7: Cant connect to AK8963

tomasbrincil, 2019-07-07:
Hi there,
I have this output from Console, can you check it before I start reading the code where could be the problem?
```
MPU9250 WHO AM I = 73
Could not connect to MPU9250: 0x0
I2C ERROR CODE : 2
ax = -1687.50 ay = -452.15 az = -539.55 mg
gx = -5.92 gy = 2.52 gz = 0.02 deg/s
mx = 0 my = 0 mz = 0 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
< calibration parameters >
accel bias [g]: 
0.00, 0.00, 0.00
gyro bias [deg/s]: 
0.00, 0.00, 0.00
mag bias [mG]: 
0.00, 0.00, 0.00
mag scale []: 
1.00, 1.00, 1.00
roll  (x-forward (north)) : 0.00
pitch (y-right (east))    : 0.00
yaw   (z-down (down))     : -7.51

```
I2C is working fine, with other code the reading is fine. 0x0C address of AK8963 is readible with this just fine and without any problem:
#define MAG_ADDRESS 0x0C

```
void I2Cread(uint8_t Address, uint8_t Register, uint8_t Nbytes, uint8_t* Data)
{
Wire.beginTransmission(Address);
Wire.write(Register);
Wire.endTransmission();
Wire.requestFrom(Address, Nbytes); 
uint8_t index=0;
while (Wire.available())
Data[index++]=Wire.read();
}

uint8_t Mag[7]; 
I2Cread(MAG_ADDRESS,0x03,7,Mag);
int16_t mx=-(Mag[3]<<8 | Mag[2]);
int16_t my=-(Mag[1]<<8 | Mag[0]);
int16_t mz=-(Mag[5]<<8 | Mag[4]);
```
I wanted use your library because I like to make a clibration simple :))
Thank you!

tomasbrincil, 2019-07-07:
Try before cry, I got it...

MPU9250.h / line: 26
```
- const uint8_t MPU9250_WHOAMI_DEFAULT_VALUE {0x71}; // 0x68????
+ const uint8_t MPU9250_WHOAMI_DEFAULT_VALUE {0x73}; // 0x68????
```

closed 2019-07-07 by tomasbrincil

sabado, 2019-08-01:
I'm having a similar issue, i changed 0x73 but cound not connect to AK8963.  Using a GY-91 breakout board.  Magnetometer data is available in another library, but i dont known why here is not working. 

I do not want to switch to 'others' library, i  really like the job you done here. 

```
MPU9250 WHO AM I = 73
MPU9250 is online...
I2C ERROR CODE : 2
AK8963  WHO AM I = 0
Could not connect to AK8963: 0x0
I2C ERROR CODE : 2
ax = 277.83 ay = 60.55 az = -965.82 mg
gx = 2.50 gy = -0.98 gz = 0.67 deg/s
mx = 0 my = 0 mz = 0 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
< calibration parameters >
accel bias [g]: 
0.00, 0.00, 0.00
gyro bias [deg/s]: 
0.00, 0.00, 0.00
mag bias [mG]: 
0.00, 0.00, 0.00
mag scale []: 
1.00, 1.00, 1.00
roll  (x-forward (north)) : 0.00
pitch (y-right (east))    : 0.00
yaw   (z-down (down))     : -7.51
```