hideakitai/MPU9250 #8: Calibration eeprom UNO Mega, how to save

duckman1961, 2019-10-04:
Hi lm trying to use the code above, but unsure how to.
I have a Mega and  MPU9255 giving live data on the sketch, it says "Calibration DONE" gives the offsets, but then it says "Calibration?: NO", then nothing else.
I replaced the 
 //  mpu.printCalibration();
    printCalibration();
It now prints the defaults stuff, but 0's
How do you get the displayed  cal data into the eeprom and also which sketch then uses the eeprom saved cal  data to give Euler's.
Thank you for the above because KrisWiners code does lack the ability to be started calibrated.

marcosroriz, 2019-10-15:
I'm also having this problem. I'm being unable to save the calibration in Arduino Mega.

zhongmc, 2020-03-08:
it seems  not completed yet; just use EEPROM to write and read; like:
loadFromEEProm:

    if (EEPROM.read(EEP_CALIB_FLAG) == 0x01)
    {
        Serial.println("calibrated to eeprom? : YES");
        Serial.println("load calibrated values");
        float value;
        for( int i=0; i<3; i++)
        {
          EEPROM.get( EEP_ACC_BIAS + 4*i,  value);
          mpu.setAccBias(i, value);
        }
    ]
......

save to EEprom:

        EEPROM.write(EEP_CALIB_FLAG, 1);
        EEPROM.put(EEP_ACC_BIAS + 0, mpu.getAccBias(0));
        EEPROM.put(EEP_ACC_BIAS + 4, mpu.getAccBias(1));
        EEPROM.put(EEP_ACC_BIAS + 8, mpu.getAccBias(2));
        EEPROM.put(EEP_GYRO_BIAS + 0, mpu.getGyroBias(0));
        EEPROM.put(EEP_GYRO_BIAS + 4, mpu.getGyroBias(1));
     ......

hideakitai, 2020-11-04:
Sorry for late reply, now [calibration_eeprom](https://github.com/hideakitai/MPU9250/tree/master/examples/calibration_eeprom) example can be also used for AVR. Thanks!

closed 2020-11-04 by hideakitai

duckman1961, 2020-11-04:
Thanks hideakitai, I appreciate you completing the examples, l won't be
home until Sunday to play with the MPU9255, but am looking forward to using
it. Thanks again.