hideakitai/MPU9250 #15: with esp8266

froussel, 2020-11-19:
I am looking for a program with an mpu9250 and an ESP8266 which would allow me to know the pitch, the roll relative to the horizontal as well as the direction of the north. 
This is to put in a model multihull sailboat and be able to check that it is going in the right direction and act on the foils according to the roll and pitch.
I understand the role of accelerometers, gyroscopes, but I'm bad at trigonometry.
For 3 days I have been looking for something that works.
Many libraries including yours.

I test the "Simple Measurement" program
with       mpu.setup(0x68); // you can set your own address and I have

I2C ERROR CODE : 4
I2C ERROR CODE : 4
I2C ERROR CODE : 4
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = 0 my = 0 mz = 0 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Yaw, Pitch, Roll: 0.00, 0.00, 0.00
< calibration parameters >
accel bias [g]: 
0.00, 0.00, 0.00
gyro bias [deg/s]: 
0.00, 0.00, 0.00
mag bias [mG]: 
0.00, 0.00, 0.00
mag scale []: 
1.00, 1.00, 1.00
roll  (x-forward (north)) : 0.00
pitch (y-right (east))    : 0.00
yaw   (z-down (down))     : 0.00

Where is my mistake please ?

hideakitai, 2020-11-19:
You got I2C error 4 (see [here](https://github.com/arduino/ArduinoCore-avr/blob/master/libraries/Wire/src/utility/twi.c#L250) for detail)
> 4 .. other twi error (lost bus arbitration, bus error, ..)

I think you have some hardware mistakes (but I don't know what it is...).

froussel, 2020-11-20:
with arduino **nano** and no problem of connection 

BBU⸮W\⸮8⸮⸮I2C ERROR CODE : 2
ax = -126.95 ay = 4162.11 az = 7121.09 mg
gx = -9.77 gy = -1.53 gz = -6.10 deg/s
mx = 0 my = 0 mz = 0 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
< calibration parameters >
accel bias [g]: 
0.00, 0.00, 0.00
gyro bias [deg/s]: 
0.00, 0.00, 0.00
mag bias [mG]: 
0.00, 0.00, 0.00
mag scale []: 
1.00, 1.00, 1.00
roll  (x-forward (north)) : 0.00
pitch (y-right (east))    : 0.00
yaw   (z-down (down))     : -7.51

hideakitai, 2020-11-20:
Ok, then please check your i2c setting. Because this library works for all Wire instance (no difference between devices), I think the problem is i2c setting. For example, you should set SDA and SCL to correct pins on ESP series.

hideakitai, 2020-11-20:
It's also better to check if i2c works with other sensors.

froussel, 2020-11-20:
My mpu9250 is [https://fr.aliexpress.com/item/4000284459476.html](https://fr.aliexpress.com/item/4000284459476.html)
and it is good with https://github.com/bolderflight/MPU9250 both an esp8266 and an nano

hideakitai, 2020-11-20:
Let me show your test code?

froussel, 2020-11-20:
https://github.com/hideakitai/MPU9250#simple-measurement
With in the same folder your file mpu9250.h and le folder mpu9250
[http://dl.free.fr/bzaPWKzzq](
http://dl.free.fr/bzaPWKzzq)  for download the zip

hideakitai, 2020-11-20:
Ok, I recommend to test the 'simple' example, though. How about i2c pins? Have you tested with correct setting?

froussel, 2020-11-20:
can you read me a little ?

hideakitai, 2020-11-21:
I don't have devices now, I'll check it somewhere next week.

hideakitai, 2020-11-25:
@froussel Sorry for late reply, I've checked on ESP8266 and worked correctly. In my setting, SDA = 4 and SCL = 5. Can you check your hardware setting again?

``` C++
#include "MPU9250.h"

MPU9250 mpu;

void setup() {
    Serial.begin(115200);

    Wire.begin(4, 5);

    delay(2000);
    mpu.setup();
}

void loop() {
    static uint32_t prev_ms = millis();
    if ((millis() - prev_ms) > 16) {
        mpu.update();
        mpu.printRollPitchYaw();

        prev_ms = millis();
    }
}
```

froussel, 2020-11-25:
serial said 
 I2C ERROR CODE : 2
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
I2C ERROR CODE : 2
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
I2C ERROR CODE : 2...

hideakitai, 2020-11-25:
The library works correctly, so I think hardware setting is not correct. Do you have pull-ups for I2Cs?

froussel, 2020-11-25:
With [Advanced_I2C.zip](https://github.com/hideakitai/MPU9250/files/5595880/Advanced_I2C.zip)
 serial said 
0.28	-1.23	-10.17	-0.00	-0.00	0.00	-66.42	39.14	82.21	24.564262angle:	-59.49
0.27	-1.22	-10.16	-0.00	0.00	0.00	-66.42	39.14	82.21	24.561266angle:	-59.49
0.28	-1.20	-10.17	-0.00	0.00	0.00	-66.42	39.14	82.21	24.555277angle:	-59.49
0.28	-1.25	-10.15	-0.00	0.00	0.00	-66.42	39.14	82.21	24.537304angle:	-59.49

hideakitai, 2020-11-25:
Ah, the version of ESP8266 may also affects the results. I've tested above on Arduino IDE 1.8.13 and esp8266 v2.7.4.

froussel, 2020-11-25:
I work with arduino 1.8.13 et 2.7.4 pour la carte ESP8266

I have no pull-up resistance how much do you want me to put on?

What asks me questions is how come your library does not work when all the ones I have found elsewhere do it without difficulty

hideakitai, 2020-11-25:
work with arduino 1.8.13 et 2.7.4 for ESP8266

I have no pull-up resistance how much do you want me to put on?

What asks me questions is how come your library does not work when all the ones I have found elsewhere do it without difficulty

Generally you should add 10k for both SDA and SCL if the wire is short enough. If you use long wire, try to use around 5k and lower.

I don't know why other libraries works, because I2C error happens, every library won't work. So one thing I can imagine is the difference of connection stability.

Pull-ups are necessary for stable communication on I2C. See https://learn.sparkfun.com/tutorials/i2c/all

froussel, 2020-11-25:
With 4.7 K between 3.3v and sda / scl : no improvement always the same serial message

hideakitai, 2020-11-25:
Can you give me a serial message with latest master's simple example? I think it works if you have correct hardware settings.

hideakitai, 2020-11-25:
Sorry, please give me a serial message of this code. Maybe it's cleaner.

> @froussel Sorry for late reply, I've checked on ESP8266 and worked correctly. In my setting, SDA = 4 and SCL = 5. Can you check your hardware setting again?
>
>
>
> ``` C++
>
> #include "MPU9250.h" [...]

froussel, 2020-11-25:
I2C ERROR CODE : 2
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
I2C ERROR CODE : 2
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
I2C ERROR CODE : 2
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
I2C ERROR CODE : 2
...

froussel, 2020-11-25:
I looked a little at your mpu9250.h even if I do not understand everything know that when a program works other than yours it addresses the mpu with 0x68

hideakitai, 2020-11-26:
Ah... yes, 0x68 is default address for MPU9250 and you can change it to your own as I wrote in README. You should set your own address if you use another one currently.

> You can set your own address based on A0, A1, A2 setting as:
> mpu.setup(0x70); // you can set your own address

hideakitai, 2020-11-26:
@froussel I've made two changes in v0.2.1. First, `connection_check.ino` example is added and you can check if the connection is correct and can see which address is on I2C.

Second, I've changed `setup()` API to force to set your own I2C address explicitly. So everyone should understand the address setting of MPU and set it explicitly.

froussel, 2020-11-26:
connection_check.ino

Searching for i2c devices...
Found 1 I2C devices
I2C addresses are: 0x68 
I2C address 0x68 is MPU9255 and ready to use

 But my gy-65 is not read

With this scanner 👍 
`// --------------------------------------
// i2c_scanner
//
// Version 1
//    This program (or code that looks like it)
//    can be found in many places.
//    For example on the Arduino.cc forum.
//    The original author is not know.
// Version 2, Juni 2012, Using Arduino 1.0.1
//     Adapted to be as simple as possible by Arduino.cc user Krodal
// Version 3, Feb 26  2013
//    V3 by louarnold
// Version 4, March 3, 2013, Using Arduino 1.0.3
//    by Arduino.cc user Krodal.
//    Changes by louarnold removed.
//    Scanning addresses changed from 0...127 to 1...119,
//    according to the i2c scanner by Nick Gammon
//    http://www.gammon.com.au/forum/?id=10896
// Version 5, March 28, 2013
//    As version 4, but address scans now to 127.
//    A sensor seems to use address 120.
// Version 6, November 27, 2015.
//    Added waiting for the Leonardo serial communication.
//
//
// This sketch tests the standard 7-bit addresses
// Devices with higher bit address might not be seen properly.
//
#include <Wire.h>
void setup()
{
  Wire.begin();
  Serial.begin(115200);
  while (!Serial);             // Leonardo: wait for serial monitor
  Serial.println("\nI2C Scanner");
}
 
void loop()
{
  byte error, address;
  int nDevices;
  Serial.println("Scanning...");
  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");
      nDevices++;
    }
    else if (error==4)
    {
      Serial.print("Unknown error at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");
  delay(5000);           // wait 5 seconds for next scan
}`

Serial said 

Scanning...
I2C device found at address 0x68  !
I2C device found at address 0x77  !
done

New simple.ino  
Always :

I2C ERROR CODE : 2
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
....

hideakitai, 2020-11-26:
> But my gy-65 is not read
Yes, `connection_check.ino` searches only 0x68 to 0x6F which can be MPU-9250's address.

> I2C address 0x68 is MPU9255 and ready to use
Your device is not MPU9250, so please use `MPU9255 mpu;` instead of `MPU9250 mpu;` as I wrote in README.
> To use MPU9255 instead of MPU9250, just declare MPU9255.
> MPU9255 mpu;

froussel, 2020-11-26:
Great by putting 9255 instead of 9250.
Thanks !

As you know how to recognize what kind of mpu is used why not integrate it and have it done automatically?
are the pitch, roll and yaw obtained in series with simple.ino the mixture of accelerometers and gyroscopes ?.
I would like to use this IMU to act on the balance of a scale model sailing radio-controlled multihull.
So as soon as the roll and pitch are no longer correct act on 3 servos: one on a port foil, another on one on starboard and one on the rudder.
The calibration should only be done on land on a support and be memorized;

hideakitai, 2020-11-27:
The library has improved, thanks.

> As you know how to recognize what kind of mpu is used why not integrate it and have it done automatically?

Because MPU9255 has additional feature called [Automatic Activity Recognition](https://sureshjoshi.com/embedded/invensense-imus-what-to-know/), so I've chosen to separate the class. Though it's not implemented and maybe not required currently. Anyway, it's good to know what chip you are using and understand about it.

> are the pitch, roll and yaw obtained in Serial.print with simple.ino the mixture of accelerometers and gyroscopes ?.

Yes

> The calibration should only be done on land on a support and be memorized;

See the [calibration_eeprom](https://github.com/hideakitai/MPU9250/tree/master/examples/calibration_eeprom) example.

froussel, 2020-11-27:
Thank you so much.
There are just two examples that could be added to complete the use of your library:
the temperature output (very easy, even I can do it)
display of the direction of geographic north with compensation for the declination of the location taking into account pitch and roll

hideakitai, 2020-11-27:
Ah, ok I'll add temperature output. Please give me PR for the north direction!

froussel, 2020-11-27:
PR : ???

hideakitai, 2020-11-28:
It's Pull Request;)

hideakitai, 2020-11-28:
Added temperature. Now I'm going to close this issue because the issue has solved. But feel free to open new one for north direction.

closed 2020-11-28 by hideakitai