hideakitai/MPU9250 #16: printRollPitchYaw() is not refreshing, printing same values over and over in Example sketch

shivam5594, 2020-11-28:
I am trying the example code called "Simple" on my ESP32-Cam board. When I run the code as is just by editing

```
Wire.begin(15,14);
```
to my need, it keeps printing same values over and over and doesn't change even when I move the IMU.

```
ts Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00

```

What wrong am I doing?

hideakitai, 2020-11-28:
Hi, your esp is not connected to mpu maybe. Please check your connection by [connection_check example](https://github.com/hideakitai/MPU9250/blob/master/examples/connection_check/connection_check.ino) first.

shivam5594, 2020-12-03:
@hideakitai I tried that example and it detects the device at 0x68 as expected. This is the output.
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Searching for i2c devices...
Found 1 I2C devices
I2C addresses are: 0x68 
I2C address 0x68 is MPU9255 and ready to use
```
And then I used the simple example and output is still same...
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00
Yaw, Pitch, Roll: -7.51, 0.00, 0.00

```

shivam5594, 2020-12-03:
this is my code
```
#include "MPU9250.h"

MPU9250 mpu;

void setup() {
    Serial.begin(115200);
    Wire.begin(15,14);
    delay(2000);

    mpu.setup(0x68);  // change to your own address
}

void loop() {
    if (mpu.update()) {
        mpu.printRollPitchYaw();
    }
}
```

hideakitai, 2020-12-03:
> I2C address 0x68 is MPU9255 and ready to use

Your device is not MPU9250, but MPU9255.
Please try to change your declaration from `MPU9250 mpu;` to `MPU9255 mpu;`.

shivam5594, 2020-12-03:
ahh thank you.

closed 2020-12-03 by shivam5594