hideakitai/MPU9250 #17: Have an example giving the heading whatever the position of the sensor

froussel, 2020-11-28:
Since there is an example of calculating pitch and roll by mixing data from accelerometers and gyroscopes, an example of merging magnetometer data to have the direction of magnetic north, of geographic north (leaving the possibility of entering the value of declination by the user) expressed in degrees but also in vector form. Instead of giving the theta cap as of an integer from 0 to 360, it is returned in the form of a heading vector of norm 1 and coordinates (cosine (theta), sine (theta)). The sine and cosine functions being 2π periodic.

hideakitai, 2020-11-28:
Now our yaw value points out geometric north

https://github.com/hideakitai/MPU9250/blob/8e6971db748a6421cf4f46244b45dcccfa8ba339/MPU9250.h#L155

and considering magnetic declination

https://github.com/hideakitai/MPU9250/blob/8e6971db748a6421cf4f46244b45dcccfa8ba339/MPU9250.h#L437

and I have the method to set magnetic declination by degree

https://github.com/hideakitai/MPU9250/blob/8e6971db748a6421cf4f46244b45dcccfa8ba339/MPU9250.h#L240

Isn't it enough?

froussel, 2020-11-28:
See issue 18, I can't test

froussel, 2020-11-28:
After a little cleaning in my welds and
In MPU9250.h line 290 I put Serial.print
In eeprom_utils.h: extern MPU9255 mpu;

Tested : [mon_test.zip](https://github.com/hideakitai/MPU9250/files/5610665/mon_test.zip)

I have some questions in my-test.ino between / * and * /

Then the yaw varies from -180 to 180 and does not seem to indicate north

On the other hand, no problems for pitch and roll

hideakitai, 2020-11-29:
- You must calibrate acc gyro mag correctly to get correct direction
- Especially mag calibration is affected by environment, do it in open and no-electromagnetic wave
- I recommend to use `Tools > Auto Format` to keep your code arranged...
- Let me know how many degree is different and what you compared with (something maybe shows correct direction)

froussel, 2020-11-29:
Yes if 

`#include "MPU9250.h"
#include "eeprom_utils.h"
MPU9255 mpu;                        //  ou 9250 suivant le capteur utilisé
void setup() {
  Serial.begin(115200);
  Wire.begin();
  delay(2000);
  mpu.setup(0x68);                // change to your own address
  Serial.println(" début dans 2 secondes");
  mpu.calibrateAccelGyro();
  Serial.println(" fin accéléro");
  delay(2000);
  Serial.println(" début des huits en l'air");
  mpu.calibrateMag();

  /* `

You have to break down the 2 calibrations :
 - accelerometer and gyro: flat and still
 - Then mag. doing eights in the air

Would it be possible to separate the backup of these two calibrations separately? Please.
 and load also

With

mpu.SelfTest();               

Said

    In file included from C:\Users\admin\Desktop\hideakitai\mon_test\mon_test.ino:2:0:
    sketch\MPU9250.h: In function 'void setup()':
    MPU9250.h:802:10: error: 'void MPU9250_<WireType, WHO_AM_I, AFSSEL, GFSSEL, MFSSEL>::SelfTest() [with WireType = TwoWire; unsigned char WHO_AM_I = 115u; AFS AFSSEL = (AFS)3; GFS GFSSEL = (GFS)3; MFS MFSSEL = (MFS)1]' is private
    void SelfTest()  // Should return percent deviation from factory trim values, +/- 14 or less deviation is a pass
        ^
    mon_test:11:17: error: within this context
    mpu.SelfTest();                
               ^
    exit status 1
    'void MPU9250_<WireType, WHO_AM_I, AFSSEL, GFSSEL, MFSSEL>::SelfTest() [with WireType = TwoWire; unsigned char WHO_AM_I = 115u; AFS AFSSEL = (AFS)3; GFS GFSSEL = (GFS)3; MFS MFSSEL = (MFS)1]' is private

Where is my mistake?

hideakitai, 2020-11-29:
> You have to break down the 2 calibrations :
> accelerometer and gyro: flat and still
> Then mag. doing eights in the air

This library has already separated two calibrations as you see.

> Would it be possible to separate the backup of these two calibrations ? Please.

Of course you can. But it's not a library's function and you can do that separately by yourself based on  `calibration_eeprom` example. Separate its implementation by yourself.

> mpu.SelfTest();

Currently it is private function, so please test it by exposing this function to public. I'll do it public later.

hideakitai, 2020-12-21:
I've published `bool selftest()`, so please use it if you want.

closed 2020-12-21 by hideakitai