hideakitai/MPU9250 #18: Something is wrong

froussel, 2020-11-28:
With to day MPU9250.h connection_check.ino with my mpu9255 said 

 Found 0 I2C devices
I2C addresses are: 
No device found on I2C bus. Please check your hardware connection

and 
#include "MPU9250.h"
#include "eeprom_utils.h"
MPU9255 mpu;                        //  ou 9250 suivant le capteur utilisé

IDE said 

mon_test:4:9: error: conflicting declaration 'MPU9255 mpu'
 MPU9255 mpu;                        //  ou 9250 suivant le capteur utilisé
         ^
In file included from C:\Users\admin\Desktop\hideakitai\mon_test\mon_test.ino:3:0:
eeprom_utils.h:5:16: error: 'mpu' has a previous declaration as 'MPU9250 mpu'
 extern MPU9250 mpu;
                ^
exit status 1
conflicting declaration 'MPU9255 mpu'

hideakitai, 2020-11-28:
First, v0.2.3 has no difference about `connection_check.ino` from v0.2.1. Please check your hardware connection again. Or please check if it works with v0.2.1 again.

Second, as you see in the error message, the declaration of `mpu` is different in `eeprom_utils.h`. It is `MPU9250` not `MPU9255`. Are you testing your own code based on `calibration_eeprom`? Please read your error message carefully. Please, please check your code carefully.

froussel, 2020-11-28:
Sorry, a thread hanging out where it shouldn't.
I resume my test

closed 2020-11-28 by froussel