hideakitai/MPU9250 #20: Math question

g-yvr, 2020-12-06:
I currently do not have hardware to test, so reading the code in order to possibly use it in the future. 

in    QuaternionFilter.h  line 202/209
 // Integrate rate of change of quaternion
        pa = q2;
        pb = q3;
        pc = q4;
        q1 = q1 + (-q2 * gx - q3 * gy - q4 * gz) * (0.5f * deltat);
        q2 = pa + (q1 * gx + pb * gz - pc * gy) * (0.5f * deltat);
        q3 = pb + (q1 * gy - pa * gz + pc * gx) * (0.5f * deltat);
        q4 = pc + (q1 * gz + pa * gy - pb * gx) * (0.5f * deltat);

When calculating q2, q3, and q4, q1 as already been changed. I think (q1, pa, pb, pc)  is not a valid quaternion, and would lead to (probably small) error. I think moving it as the last of the 4 updates, and replacing the qx by their saved p values would solve it.  

Thanks for commenting.

hideakitai, 2020-12-21:
Hi, it may be so, but sorry I can' t check it now. It's based on [this](https://github.com/kriswiner/MPU9250/blob/master/quaternionFilters.ino) library, and I'm happy if you create the PR for this.  Thanks!

hideakitai, 2020-12-24:
I'm considering to improve math equations together with #21.

hideakitai, 2020-12-25:
I'll refactor madgwick filter later, so this would be done in near future

closed 2020-12-25 by hideakitai

hideakitai, 2020-12-25:
In the latest commit, madgwick filter has improved maybe. Please check if it works correctly.

g-yvr, 2020-12-26:
Thank you very much. As mentioned before I do not have the necessary HW to run real life test. Anyhow i very much appreciate that you take my comment into account, and believe the modification is a real improvement that will benefit everyone who follows your publication. 
Best regards
Gerard