hideakitai/MPU9250 #21: Callibration used wrong in data request

John-Val, 2020-12-21:
Dear Hideaki Tai,
I am trying to learn more about the MPU9255 sensor and came across you library.

While testing I got better results from the library while not using the calibration offered. E.g. calibrating my IMU gave a gyro bias of -0.67 for the X-axis in the serial monitor. After that I got readings for gyroX all around + 0.67, while without calibration these readings are close around 0.0. So I dived in your code and found 
in updateAccelGyro

g[0] = (float)MPU9250Data[4] * gRes -  - gyroBias[0];

to my opnion this should be 

g[0] = ((float)MPU9250Data[4] - gyroBias[0])* gRes;

while in calibrateMPU9250(float* dest1, float* dest2) I read:

dest1[0] = (float)gyro_bias[0] / (float)gyrosensitivity;

As far as I understand, in this line you make the bias scale independent (you use  gyro full-scale to 250 degrees  in callibration )
and dest1 is set equal to gyroBias in
 void calibrateAccelGyro() {
        calibrateMPU9250(gyroBias, accelBias);
    }  

Gyro, accelleration readings suffer both.

The magnetometer readings are not affected in this way, but can changing the sensor resolution from 14 to 16 bits after calibration cause an error? 

Another question: Calibrating the magnetometer requires a figure 8 movement of the sensor. In a certain plain or in any direction?

hideakitai, 2020-12-24:
Thanks, it seems right, but I'll check it later. I'm considering to improve math equations together with #20 . Because these maths are just imported from Kris's [MPU9250](https://github.com/kriswiner/MPU9250).

John-Val, 2020-12-24:
Actually I now think it should even be only
g[0] = (float)MPU9250Data[4] * gRes;

while the bias is written back to the MPU and plain readings as above seem to be corrected by the bias.
John

hideakitai, 2020-12-24:
Yes, I think so. Please tell me if you have noticed anything else.

hideakitai, 2020-12-25:
@John-Val I've refactored terrible source codes. It looks good, but can you check if it works? I also want to confirm the calculation of madgwick filter is correct someday...

hideakitai, 2020-12-25:
@John-Val Madgwick filter has also improved maybe... Please check if it works fine!

John-Val, 2020-12-25:
Dear Hideaki,
I will look it at probably tomorrow, certainly on Monday. I do not know if Christmas is something you celebrate, but if so I wish some nice days.
John

hideakitai, 2020-12-28:
I've found that roll pitch yaw calculation was wrong, so fixed it in the latest master branch. Please check if it works on the latest commit.

John-Val, 2020-12-28:
Dear Hideaki,

First I have to mention that your code cleanup made me very happy.

In my fork https://github.com/John-Val/MPU9250 I made some modifications. I think the rpy code you added gave wrong results.

In the project I renamed quaternion.h to driftFilter.h and added two updates, one which does not filter and a mahony version only using the acceleration sensor.
in the example MPU9255_test_lib you can show how it can be used.
Ofcourse we can rename driftFilter back to quaternion if you like.

I also made some changes to MPU9250.h, I hope you agree with those

I added a start function to set the timer to current time.

I also moved the timing from quaternion.h to MPU9250.h while I had some 0.0 readings on the magnetometer while printing.

I also made changes to the print routines, separating headers from data, such that the Serial Plotter can be used for graphing.

Next I moved all quaternion buisness to driftFilter.h in order to cope with the various filters.

I have also a question about how to set up the struct MPU9250Setting in an  .ino file to pass as a second argument to mpu.setup.

hideakitai, 2020-12-29:
> I made some modifications. I think the rpy code you added gave wrong results.

Your code seems not to be based on my latest master branch. Please fork it again.

> In the project I renamed quaternion.h to driftFilter.h and added two updates, one which does not filter and a mahony version only using the acceleration sensor.
> in the example MPU9255_test_lib you [...]

Too many changes. Please devide them to several appropriate PRs.

> I have also a question about how to set up the struct MPU9250Setting in an .ino file to pass as a second argument to mpu.setup.

Please see this example https://github.com/hideakitai/MPU9250/blob/master/examples/custom_setting/custom_setting.ino

hideakitai, 2021-01-02:
This issue was converted to PRs, close this issue.

closed 2021-01-02 by hideakitai