hideakitai/MPU9250 #22: A working project until ver. 0.30

Attigliuzzo, 2020-12-26:
Hi hideakitai,
i would to say thanks to port Kris's code, yours is the only working library which gives me yaw, pich, roll. I wrote a simple code to build an headtracker with my Arduino micro+mpu9250 using the hatire plugin. [https://github.com/opentrack/opentrack/wiki/Built-a-Hatire-headtracker-3DOF-with-a-MPU9250](url)

I'm typing these 2 words to ask few things:
1) Code =<0.30 works with my project but not with 0.40. See my short video attached it uses the version 0.30, It could due to the change of original kris's code  #20 . . _Could you restore it?_

https://user-images.githubusercontent.com/42100711/103156325-d98dee00-47a7-11eb-827a-ba5f9598bcf0.mp4

2) in your calibration sketch i added these three dummy code lines:

`Serial.println("draw some spheres!!, after 5 seconds from now!!!");
  delay(5000);
 Serial.println("now!!!!");`

between mpu.calibrateAccelGyro(); and mpu.calibrateMag(); rows. because i saw an Edtracker videos where they calibrated the accel and gyro not moving it, simply the MPU9250 was on a table. After 5 seconds you can calibrate magnetometer drawing several eight or spheres.
Regards

Attilio

hideakitai, 2020-12-27:
Hi @Attigliuzzo , thanks for reporting. Version 0.4.0 is WIP now, so I'm happy debugging with you :)

> 1. Code =<0.30 works with my project but not with 0.40. See my short video attached it uses the version 0.30, It could due to the change of original kris's code #20 . . Could you restore it?

Could you tell me what is the problem? e.g. cannot compile, yaw/pitch/roll is strange, etc...
Please note that the yaw/pitch/roll order is not same as the v0.3.0 (not same as plane's direction) now.

> 2. in your calibration sketch i added these three dummy code lines:

Is this a suggestion? I also think it's better :)

hideakitai, 2020-12-27:
Please check the latest master branch again. You can switch raw direction and air-plane direction (default before v0.3.0) by latest commit.

Attigliuzzo, 2020-12-27:
> Please check the latest master branch again. You can switch raw direction and air-plane direction (default before v0.3.0) by latest commit.

Ok, I am not at the pc but let me to explain. Your code ver. 0.30 with all improvements worked as you can watch in my video. In the WIP 0.40 ver. I read, in a issue, that "you have improved the quaternions" from the madwick matrix calculations. So I think, leave all your changes but restore the 0.30 quaternions matrix calculations in the 0.40 code. I will test the WIP code with old quaternions. This is my idea

Yes, you can add my strings to help anyone to calibrate it correctly.

hideakitai, 2020-12-28:
I've checked the calculation again. Quaternion works fine, but roll pitch yaw calculation was not, so I've reverted it. Please check the latest master branch again.

>  I found the function useRawPitchYawRollDirection(). So if the boolean value is "false" i will use airplane directions (aka negative values), like older version of the library. Is it true?

Yes, the default is air-plane direction. In the WIP code you've tested first was not air-plane direction, but raw direction of sensor.

Attigliuzzo, 2020-12-28:
@hideakitai , everything seems gone back to the normality. Yehhhhh!!!!

closed 2020-12-28 by Attigliuzzo