hideakitai/MPU9250 #30: Wrong Linear Acceleration readings

Battou2501, 2021-01-15:
So I tried to get linAcc, acceleration excluding gravity, and Im getting strange results.  Its like Gravity doubled instead of substracted. When it lays flat a get x ~ -0.1, y ~0.02 and z ~2.01!!

Regular Acc readings are normal. So it appears there is an error with gravity exclusion calculation.

Battou2501, 2021-01-15:
So. I have read a31 a32 and a33 and compared them to a1 a2 and a3 in this part of code:

`    void update_euler(float qw, float qx, float qy, float qz) {
        // Define output variables from updated quaternion---these are Tait-Bryan angles, commonly used in aircraft orientation.
        // In this coordinate system, the positive z-axis is down toward Earth.
        // Yaw is the angle between Sensor x-axis and Earth magnetic North (or true North if corrected for local declination, looking down on the sensor positive yaw is counterclockwise.
        // Pitch is angle between sensor x-axis and Earth ground plane, toward the Earth is positive, up toward the sky is negative.
        // Roll is angle between sensor y-axis and Earth ground plane, y-axis up is positive roll.
        // These arise from the definition of the homogeneous rotation matrix constructed from quaternions.
        // Tait-Bryan angles as well as Euler angles are non-commutative; that is, the get the correct orientation the rotations must be
        // applied in the correct order which for this configuration is yaw, pitch, and then roll.
        // For more see http://en.wikipedia.org/wiki/Conversion_between_quaternions_and_Euler_angles which has additional links.
        float a12, a22, a31, a32, a33;  // rotation matrix coefficients for Euler angles and gravity components
        a12 = 2.0f * (qx * qy + qw * qz);
        a22 = qw * qw + qx * qx - qy * qy - qz * qz;
        a31 = 2.0f * (qw * qx + qy * qz);
        a32 = 2.0f * (qx * qz - qw * qy);
        a33 = qw * qw - qx * qx - qy * qy + qz * qz;
        euler[0] = atan2f(a31, a33);
        euler[1] = -asinf(a32);
        euler[2] = atan2f(a12, a22);
        euler[0] *= 180.0f / PI;
        euler[1] *= 180.0f / PI;
        euler[2] *= 180.0f / PI;
        euler[2] += magnetic_declination;
        if (euler[2] >= +180.f)
            euler[2] -= 360.f;
        else if (euler[2] < -180.f)
            euler[2] += 360.f;

        lin_acc[0] = a[0] + a31;
        lin_acc[1] = a[1] + a32;
        lin_acc[2] = a[2] - a33;
    }`

This is What I get when MPU is on its side (XYZ - a1,2,3; Grav - a31,32,33):

X: -0.04 --- Grav: 0.99 --- Result: 0.95
Y: -1.02 --- Grav: 0.07 --- Result: -0.95
Z: -0.06 --- Grav: 0.08 --- Result: -0.14

So a31 and a32 is applied to wrong acc axes.

Furthermore, this is what i get when it lays flat:

X: -0.08 --- Grav: 0.01 --- Result: -0.07
Y: 0.02 --- Grav: -0.07 --- Result: -0.05
Z: 1.04 --- Grav: -1.00 --- Result: 2.03

In your code you substracting a33 from a3, but substracting negative value will double acceleration.
 

And this is when its pointing up:

X: -1.01 --- Grav: 0.07 --- Result: -0.94
Y: -0.01 --- Grav: -0.99 --- Result: -1.01
Z: 0.05 --- Grav: -0.10 --- Result: 0.15

Again a31 and a32 applied to wrong axes, but also you are summing a1 with a31 and a2 with a32, so here you also will double acc.

So the right code would be:

`
        lin_acc[0] =  a[0] - a32;
        lin_acc[1] = a[1] + a31;
        lin_acc[2] = a[2] + a33;
`

hideakitai, 2021-01-16:
@Battou2501 Thanks for reporting, maybe it was broken when I've refactored this library. Could you make the PR for this?

Battou2501, 2021-01-16:
> @Battou2501 Thanks for reporting, maybe it was broken when I've refactored this library. Could you make the PR for this?

Pull Request? Honestly I dont know how, never tried. When I go to Create Pull Request the button to create is inactive.

hideakitai, 2021-01-17:
Ok, then I'll confirm if it works somewhere next week.

hideakitai, 2021-05-24:
@Battou2501 This issue will be fixed by https://github.com/hideakitai/MPU9250/tree/bugfix_obromious. Please confirm if it works.

hideakitai, 2021-05-28:
Fixed in v0.4.3. Please reopen the issue if you still have the problem.

closed 2021-05-28 by hideakitai