hideakitai/MPU9250 #31: I have no yaw reading at all

bavareze, 2021-01-19:
Hi. I am running the simple.ino

As I am moving my board around each axis I can see changes in pitch and roll.  However, the yaw value shows no change at all.

So I wanted to check raw data and I noticed that again, gyrox and gyroy show values consistent with my movements, but gyroz is pretty much motionless.

Are you getting similar results?  Could my sensor board (HiLetgo) be bad?

thanks

hideakitai, 2021-01-19:
Hi, I haven't experienced such situation... Your MPU may be broken. If `connection_check.ino` example works correctly (and you choose correct device (MPU9250 or MPU9255)), you can get roll pitch yaw without such problem. Could you try another sensor board?

bavareze, 2021-01-19:
Yes. I already ordered another one. Will keep you posted. Thanks

bavareze, 2021-01-30:
Hello,

very bizarre, my yaw gyro was indeed bad.  I thought this would never
happen again since is electronic component.

but anyway.

I need your help:

my sensor is inside a box with many electronics and magnetic fields.  when
I yaw the box the gyro shows motion but then the madgwick filter "corrects"
back the yaw position because the magnetic field has moved too.

So that means I need to change the madgwick filter to use everything except
for magnetometer (with the risk of yaw drift over time, that's ok)

Any idea how to do that

Thanks!

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

hideakitai, 2021-02-03:
Hi, you can just read the raw accel/gyro/mag data, and you can apply madgwick filter by yourself. The [update() function](https://github.com/hideakitai/MPU9250/blob/0929da613d362cb546767d42566d78c47c1a473e/MPU9250.h#L187) shows how to do that.

hideakitai, 2021-02-24:
I'll close this issue but please feel free to reopen it.

closed 2021-02-24 by hideakitai

bavareze, 2021-02-24:
gotcha thanks

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>