hideakitai/MPU9250 #32: Mpu 9250 gyro z axies not working.

HubertPiatkowski, 2021-02-01:
Dear Hidea Kitai,
do you have any idea why Z axis data from mpu 9250 has values oscilating around 0, when rotating board along Z axies? X and Z axies data are fine. I try it on two boards with mpu 9250 already with same result.

hideakitai, 2021-02-03:
Hi, which data do you read? 
First, please try `connection_check` example to verify your connection.
Second, read the raw accel/gyro/magneto data. If you can read it, maybe it's caused by the failure of the device. 
Third, read the Euler angle or roll/pitch/yaw and if you can't read it correctly, it's the failure of calibration.

HubertPiatkowski, 2021-02-03:
All raw readings are valid, except gyro Z.  

I contacted with the seller. He told that production of 9025 is anded. So most of mpu 9250 on market are reused (like seckend hand), he said. So most of cheeper ones will have malafuction like mine mpu board. He told me that he will check all mpu 9250 that he has on stock.

Is that possible, are cheep mpu 9250 mostly reused?

hideakitai, 2021-02-24:
Sorry for late reply. Yes, I think the IC can be reused, but it sometimes destroys the IC because of the heat when removing it. I will close this issue, I hope it works out for you.

closed 2021-02-24 by hideakitai