hideakitai/MPU9250 #34: Gyro readings noisy and do not respond to low pass filter bandwidth changes

obromios, 2021-02-18:
There may be a simple mistake in the library that is causing very noise Gyro readings for some settings.

If I use the default settings for this library, the Low Pass Filter (LPF) on the Gyro should be 41Hz.  When I look at the calibrated gyro readings they are very noisy e.g.

```
-1.10, -15.69, 1.40
-0.67, -1.59, -4.82
-1.71, -3.78, 5.13
0.31, -6.04, 0.49
-1.10, 17.33, 8.36
```
If I calculate the average of the total RMS error it is about 6.0 degrees per second.  If you look at Table 1 of the MPU 9250 Product Specification (Revision 1.1), it states that at a LPF bandwidth of 92 Hz, the total RMS error should be less that 0.1 degrees per second, i.e. the measured RMS error is about 60 times higher than specification.  [I am assuming that the Total  RMS error is the square root of the sum of the squares of the x, y, and z gyro measurements, and that  the calibration error is zero].

I tried setting the LPG filter to 92Hz but it had little effect on the RMS error, and in general it had no effect, even setting it as high as 3600. However the bandwidth should have a strong effect on the RMS error.

Looking at the code, I think the problem is in ```initMPU9250()```. If you look at [line 346](https://github.com/hideakitai/MPU9250/blob/master/MPU9250.h#L346), it loads the Fchoice setting into bits 0:1 of the ```GYRO_CONFIG``` register.  The default fchoice setting for the library is  ```0x03```  or in binary ```11```.  The difficulty arises because bits 0:1 of the register are not Fchoice but Fchoice_b. As noted in table 7 of[ MPU-9250 Register Map ](https://longnight975551865.wordpress.com/2018/02/11/how-to-read-data-from-mpu9250/) (Revision 1.6),  Fchoice_b is the bitwise inversion (twos complement) of F_choice. This can be seen in this comment at line 883 in Kris Winers [original code](https://github.com/kriswiner/MPU9250/blob/master/MPU9250_MS5637_AHRS_t3.ino#L883):
```
 // c =| 0x00; // Set Fchoice for the gyro to 11 by writing its inverse to bits 1:0 of GYRO_CONFIG
```
This problem can be easily fixed by changing [line 346](https://github.com/hideakitai/MPU9250/blob/master/MPU9250.h#L346) to
```
 c = c | (uint8_t(~setting.gyro_fchoice) & 0x03);       // Set Fchoice for the gyro by loading complement of Fchoice into Fchoice_b bits.
```
i.e use the complement of setting.gyro_fchoice.

An alternative workaround is the use the complement of the require Fchoice in the settings e.g.
```
setting.gyro_fchoice = 0x00
```
will allow the LPF bandwidth settings to work.

With this  change, the measurements are far less noisy. The total RMS error at 92 HZ drops from 6 degrees per second to 0.06 degrees per second, so within the  specification of 0.1 degrees per second.

hideakitai, 2021-02-24:
I hope you will make the pull request for this (#33). Thank you for your contribution :)

obromios, 2021-02-25:
I  will include it in the pull request.

hideakitai, 2021-05-24:
@hideakitai This issue will be fixed by https://github.com/hideakitai/MPU9250/tree/bugfix_obromious. Please confirm if it works.

obromios, 2021-05-28:
@hideakitai, I have checked and the gryo readings appear to be correct on this [branch](https://github.com/hideakitai/MPU9250/tree/bugfix_obromious).

hideakitai, 2021-05-28:
Fixed in v0.4.3

closed 2021-05-28 by hideakitai

Hantuch, 2021-06-03:
Sorry if i'm wrong but should've this change be applied to the accel_fchoice_b too? 

i checked at last release and it's still

>` c = c | ((uint8_t(setting.accel_fchoice) & 0x01) << 3);  // Set accel_fchoice_b to 1`

hideakitai, 2021-06-04:
@Hantuch Yes, it should be inverted... I will fix it, thanks!

obromios, 2021-06-04:
I have not got time to check until  next week, but my recollection is that
I thought the same thing, but when I checked the spec, it was actually
correct for the accelerometer, i.e. it should not be inverted.

Chris

hideakitai, 2021-06-04:
@obromios I think it should be inverted based on the spec sheet...

[image]

Hantuch, 2021-06-04:
@obromios does the accelerometer output was correct when you checked it? i don't have the means to check it

@hideakitai yeah that's what i checked too, it's also the same with MPU6500

[image]

obromios, 2021-06-07:
Having checked, I  agree with @Hantuch and @hideakitai, that the ACCEL_FCHOICE_B bit in the register is inverted version of ACCEL_FCHOICE, and currently in the code there is no apparent inversion of ```setting.accel_fs_sel```.

On the other hand, if I do ```settings.accel_fchoice = 0x01```,  and ```settings.accel_dlpf_cfg = ACCEL_DLPF_CFG::DLPF_99HZ``` the RMS value of the accelerometer readings are about 7.5 millig, which agrees with the spec that states the maximum RMS value should be 8 millig at 92 Hz. If I set  ```settings.accel_fchoice = 0x00```, then the RMS value is about 450mg, which makes sense if there is no filtering at all.  Maybe I am not thinking about this right, but it looks like the accel_fchoice is working correctly, but it looks like the wrong value is being loaded into the register.

Another puzzling thing is that if I set ```settings.accel_fchoice = 0x01``` and change the filter to ```settings.accel_dlpf_cfg = ACCEL_DLPF_CFG::DLPF_420HZ``` the RMS value is still about 7.5millig, whereas I would have expected it to be about twice as large.  Maybe someone else can check these settings and my calculations to see if they are seeing similar results.

Note that to calculate RMS readings I am multiplying the ```getAccX/Y/Z``` by 1000 to convert from g to millig, so I can check directly against the specification.  The code I am using to do the RMS measurements is 

`
double n_mu {0};
double mu[3] {0};

void print_acc_rms(){
  if (mpu.update()) {
    static uint32_t prev_ms = millis();
    if (millis() > prev_ms + 25) {
      double acc_ary[3] {mpu.getAccX() * 1000, mpu.getAccY() * 1000, mpu.getAccZ() * 1000};
      n_mu += 1;
      double r = rms_acc(acc_ary, mu);
      moving_mu_r = (r + (n_mu - 1) * moving_mu_r) / n_mu;
      Serial.print("moving_rms: ");Serial.print(", ");Serial.print(moving_mu_r, 3);
      Serial.println();
      prev_ms = millis();
    }
  }
}

double rms_acc(double ary[3], double ary_mu[3]){
  double mean_square = (sq(ary[0] - ary_mu[0]) + sq(ary[1] - ary_mu[1]) + sq(ary[2] - ary_mu[2] - 1000.0)) / 3.0;
  return sqrt(mean_square);
}
` 
@hideakitai you might like to consider having ```getAccX``` etc output in milligs so the output aligns with the  specification. This change should not be a breaking change because I believe the filters use normalised values.

obromios, 2021-06-07:
I founds some bugs in my code to calculate the RMS. When I fixed this,  the RMS readings make sense and are in accord with the need to invert the Accelerometer f_choice.  With 
`  settings.accel_fchoice = 0x00;
  settings.accel_dlpf_cfg = ACCEL_DLPF_CFG::DLPF_99HZ;`
I get an RMS of 1.9 millig, well within the specification. If I set
`  settings.accel_fchoice = 0x01;
  settings.accel_dlpf_cfg = ACCEL_DLPF_CFG::DLPF_99HZ;`
The RMS goes to 5.2 millig, i.e. the 99HZ filter has been disabled.

With the ```settings.accel_fchoice = 0x00``` , if I vary the filter bandwidth from 99 HZ down to 5 Hz, the RMS error reduces in a consistent fashion (though  a little slower than the sqrt of bandwidth relationship that I would have expected).

Thanks to @Hantuch for finding this fault.

The only strange results I saw is that if I set 
`  settings.accel_fchoice = 0x00;
  settings.accel_dlpf_cfg = ACCEL_DLPF_CFG::DLPF_420HZ;`
then the mean of the z measurement dropped to about 250 millig.  In all In all settings the mean of the x and y measurements were about zero and the mean of the z measurement was about 1000, as expected.

 Can anyone duplicate this?  Does anyone have any explanation why this might be?  

The corrected code I used for the RMS measurements is

```
double moving_mu_r {0};  // setup moving averages
double n_mu {0};
double mu[3] {0};

void print_acc_rms(){
  if (mpu.update()) {
    static uint32_t prev_ms = millis();
    if (millis() > prev_ms + 25) {
      double acc_ary[3] {mpu.getAccX() * 1000, mpu.getAccY() * 1000, mpu.getAccZ() * 1000};
      n_mu += 1;
      for (int i = 0; i < 3; ++i)
      {
        mu[i] = (acc_ary[i] + (n_mu - 1) * mu[i]) / n_mu;
      }
      double r = rms(acc_ary, mu);
      moving_mu_r = (r + (n_mu - 1) * moving_mu_r) / n_mu;
      Serial.print("readings: ");Serial.print(acc_ary[0]);Serial.print(", ");Serial.print(acc_ary[1]);Serial.print(", ");Serial.println(acc_ary[2]);
      Serial.print("moving_rms: ");Serial.print(", ");Serial.print(moving_mu_r, 3);
      Serial.println();
      prev_ms = millis();
    }
  }
}

double rms(double ary[3], double ary_mu[3]){
  double mean_square = (sq(ary[0] - ary_mu[0]) + sq(ary[1] - ary_mu[1]) + sq(ary[2] - ary_mu[2])) / 3.0;
  return sqrt(mean_square);
}
```

hideakitai, 2021-06-08:
@obromios If ACCEL_FCHOICE is 0, ACCEL_DLPF_CFG is "don't care". Do you mean only the DLPF_420Hz affects the result?

Hantuch, 2021-06-08:
Is there any chance the 4KHz output at ACCEL_FCHOICE=0 /ACCEL_FCHOICE_B=1 affecting the code or the i2c?

obromios, 2021-06-09:
@hideakitai, my setting of ```settings.accel_fchoice = 0x00``` is with the current code base, so it is directly loaded into the register, so I think it sets ```ACCEL_FCHOICE_B=0```, so the filters should work  (which they do).

@Hantuch I have not looked at the sample rate divider and am just using whatever the default setting is.  It is possible a 4KHz output is somehow affecting my devices, so it would be good if someone else can test all this on their circuit.

obromios, 2021-06-13:
@hideakitai, just to be clear, I think you should invert at
`c = c | ((uint8_t(setting.accel_fchoice) & 0x01) << 3); // Set accel_fchoice_b to 1`
The problem  with the 420Hz low pass filter may be an artifact of my system.

hideakitai, 2021-06-13:
@obromios Do you mean here?

https://github.com/hideakitai/MPU9250/blob/df57fbaee24c74dda65902044ae58bcc9c44df58/MPU9250.h#L382

I think it is already fixed in the latest version (v0.4.4). Could you test in the latest version?

obromios, 2021-06-13:
Excellent. I will test it sometime this week.

obromios, 2021-06-17:
I have tested the latest version and it works well.  Between this change and the similar change to the gryo fchoice, there should now be a very considerable improvement to the pitch and roll angles. I am continuing to work to improve the mag calibration.