hideakitai/MPU9250 #35: Definition of setMagneticDeclination() method

sebvaillancourt3, 2021-02-24:
This is less of an issue and more of a question. In your repo., there is no .cpp file with the definition of methods for class MPU9250. I'd like to know the inner workings of this method and how this magnetic declination is applied on the raw measurements to get correct yaw reading. Can you tell me where I can find the class/methods definition?

hideakitai, 2021-02-24:
Definition is here:

https://github.com/hideakitai/MPU9250/blob/0929da613d362cb546767d42566d78c47c1a473e/MPU9250.h#L301

and it is used here:

https://github.com/hideakitai/MPU9250/blob/0929da613d362cb546767d42566d78c47c1a473e/MPU9250.h#L429

sebvaillancourt3, 2021-02-24:
wonderful ok. Now I know the magnetic declination is actually the yaw declination and not the pitch declination that accounts for the geomagnetic field dip. The whole code seems to operate very well when the sensor is held flat, but as soon as the pitch or  change the yaw goes crazy. Any idea how to counter this effect?

hideakitai, 2021-02-24:
I think the calibration haven't done well. You should calibrate the IC correctly to get the better performance. Please try the calibration again and if you still have the problem, please feel free to let me know.

closed 2021-02-24 by hideakitai