hideakitai/MPU9250 #36: MPU connection failed. Please check your connection with `connection_check` example.

phonecol, 2021-03-04:
when i upload the simple.ino, the serial monitor will message this MPU connection failed. Please check your connection with `connection_check` example.. I dont know whats wrong with this

hideakitai, 2021-03-04:
It means exactly as it is written. First check the hardware connection with `connection_check` example.
 https://github.com/hideakitai/MPU9250/blob/master/examples/connection_check/connection_check.ino

GusakIurii, 2021-03-05:
I also get an error:
I2C ERROR CODE: 2
MPU connection failed. Please check your connection with `connection_check` example.

When compiling the `connection_check` sketch, the error is:
connection_check: 7: error: variable or field 'scan_mpu' declared void
connection_check: 7: error: 'WireType' was not declared in this scope
connection_check: 7: error: 'wire' was not declared in this scope

On another library (https://github.com/bolderflight/MPU9250), this sensor is seen without errors, and gives the coordinates.

hideakitai, 2021-03-05:
@GusakIurii What kind of board do you use? Please tell me your environment (e.g. IDE, IDE version, board, board version, etc).

GusakIurii, 2021-03-05:
https://www.banggood.com/MPU-9250-GY-9250-9-Axis-Sensor-Module-I2C-SPI-Communication-Board-Accelerometer-p-1227241.html?gmcCountry=UA&currency=UAH&cur_warehouse=CN&createTmp=1&utm_source=googleshopping&utm_medium=cpc_bgs&utm_content=haosen&utm_campaign=haosen-ssc-ua-en-all-0204&ad_id=496617779448&gclid=Cj0KCQiAyoeCBhCTARIsAOfpKxg7xDlOaV53g41RPiJEQ3X8GkJnL23dzC8rwhLvgak8Ynwv_c5SNKEaAoHYEALw_wcB - board
Arduino IDE 1.8.4

GusakIurii, 2021-03-05:
[This library ](https://github.com/bolderflight/mpu9250-arduino) is the only one that works with my hardware! But I would also like to get the coordinates of the direction of the sensor not in the axes, but in the tilt angle, and the second from the Magnetometer as the direction where it is looking.
I thought it was implemented in this library, but alas .

hideakitai, 2021-03-06:
@GusakIurii To compile for Loraduino, we should choose the board "Arduino Pro Mini ATMega328P 3.3V/8MHz", right? There was no error and I could compile `connection_check` for Loraduino successfully.

GusakIurii, 2021-03-06:
"Arduino Pro Mini ATMega328P 3.3V/8MHz" - Yes, this is the board I choose.
[image]

hideakitai, 2021-03-06:
@GusakIurii 

> I could compile connection_check for Loraduino successfully on Arduino IDE 1.8.13 (AVR Boards 1.8.3). Please use the latest version.

Please use the latest Arduino IDE version (1.8.13).

StrikeNeon, 2021-03-09:
I have a similar issue on nano33 ble (nRF52840 ) when trying to connect two sensors at the same time, connection check identifies both of them correctly, but when trying to setup the sensors i get an I2C error 2 when data ready interrupt is enabled (https://github.com/hideakitai/MPU9250/blob/master/MPU9250.h#L380)

StrikeNeon, 2021-03-09:
@hideakitai IDE is arduino IDE 1.8.13, I2C Error is 2, so NACK on transmit of address
I'm sure it's not a hardware problem on the sensor end, as both of them work when it's the only sensor connected.

`#include "MPU9250.h"

MPU9250 mpu;
MPU9250 mpu_2;

void setup() {
  pinMode(6, OUTPUT);
  digitalWrite(6, HIGH);
  while (!Serial) {}
  Serial.begin(115200);
  Wire.begin();
  delay(2000);

  if (!mpu.setup(0x68)) {  // setup mpu 1
    while (1) {
      Serial.println("ERROR WITH MPU 1 SETUP");
      delay(5000);
    }
  }
  delay(2000);
  if (!mpu_2.setup(0x69)) {  // setup mpu 2
    while (1) {
      Serial.println("ERROR WITH MPU 2 SETUP");
      delay(5000);
    }
  }
}

void loop() {
  if (mpu.update()) {
    static uint32_t prev_ms = millis();
    if (millis() > prev_ms + 25) {
      print_roll_pitch_yaw();
      prev_ms = millis();
    }
  }
}

void print_roll_pitch_yaw() {
  Serial.print("Yaw, Pitch, Roll: ");
  Serial.print(mpu.getYaw(), 2);
  Serial.print(", ");
  Serial.print(mpu.getPitch(), 2);
  Serial.print(", ");
  Serial.println(mpu.getRoll(), 2);
}`

StrikeNeon, 2021-03-09:
@hideakitai Yes, the error occurs in setup function, upon running mpu.setup method, loop section is irrelevant, as I cannot get both of them setup proper.
Worth noting that this particular behavior (MPU 1 setup success - MPU 2 fails) is seen only after the sketch in uploaded, if i reset the board then I2C Error 2 happens when trying to setup MPU 1

hideakitai, 2021-03-10:
@StrikeNeon Sorry, I forgot the important thing. MPU9250's I2C address can be changed, but you can't change AK8963's I2C address and it's fixed to 0x0C. So you can't connect to multiple MPU9250 on the same I2C bus.

If you want to, try to use I2C multiplexer or to use I2C and SPI at the same time. Please refer https://arduino.stackexchange.com/questions/19935/how-to-use-multiple-mpu9250-to-arduino-lilypad

StrikeNeon, 2021-03-12:
@hideakitai oh, well, nevermind, i got it all running via spi anyway

closed 2021-03-12 by hideakitai