hideakitai/MPU9250 #39: initMPU9250() is not initialising the accelerometer offset registers

obromios, 2021-03-26:
I found that every time a I uploaded a program, it appeared that a constant amount was being added to each of the accelerometer readings, the constant amount was different for each axis.

I found the problem was that at the start of the program I am using the ```loadCalibration()``` function from the ```eeprom_utils.h``` to set the gyro, accelerometer, and magnetometer biases.  This calls ```setAccBias()``` which in turn calls ```write_accel_offset()``` which pushes the biases to the accelerometer offset register.  It appears that ```initMPU9250() ``` does not reset these registers, so each time ```loadCalibration()``` is called, it adds to offset register. For example, if the x-axis bias is 30, then the first time the register is set to 30. But the next time it is set to 60, and so on. You can see after 10 to 20 uploads, the accelerometer readings become unusable.

The way I fixed this was to add a hardware reset to ```initMPU9250()```. Here is the modified code snippet with the added code 
being aligned hard against the left margin.
```
   void initMPU9250() {
        acc_resolution = get_acc_resolution(setting.accel_fs_sel);
        gyro_resolution = get_gyro_resolution(setting.gyro_fs_sel);
        mag_resolution = get_mag_resolution(setting.mag_output_bits);
        
write_byte(MPU9250_ADDRESS, PWR_MGMT_1, 0x80);  // Write a one to bit 7 reset bit; toggle reset device
delay(100);

        // wake up device
        write_byte(MPU9250_ADDRESS, PWR_MGMT_1, 0x00);  // Clear sleep mode bit (6), enable all sensors
        delay(100);  
```
I am not sure if this is the best way to fix this, but it appears to work and I have not yet seen any side effects.

The way most people would have experienced this problem is that for the first upload after the device is switched on, it works well, but it will fairly rapidly degrade with each upload so that eventually it no longer gives useful angle readings.  The work around is to turn the device off and then on again, which clears all the accelerometer offset register.  Another possible workaround would be to somehow only do ```loadCalibration()``` when the device is turn on.

obromios, 2021-03-26:
@hideakitai, you should be aware that I am investigating a potential[ issue ](https://github.com/kriswiner/MPU9250/issues/456) with the original Kris Winer code. This issue could affect your code as well.   I am currently doing a mathematical analysis to resolve this. If it works out that it is an issue for your code base, I will try to fix it.

Note that I still intend to do a pull request, but am just working through your codebase until I find it works satisfactorily for my application, and then will do the pull request that would aim to close the issues I have raised. Hopefully this will be in a few weeks.

hideakitai, 2021-04-02:
@obromios OK, thank you!

hideakitai, 2021-05-24:
@obromios This issue will be fixed by https://github.com/hideakitai/MPU9250/tree/bugfix_obromious. Please confirm if it works.

hideakitai, 2021-05-28:
Fixed in v0.4.3

closed 2021-05-28 by hideakitai