hideakitai/MPU9250 #42: Yaw offset problem

vollukas, 2021-04-02:
Hello,

I finally managed to get the right values for Roll and Pitch (0 on a flat surface). But no matter how many times I calibrate and how much I rotate the sensor around during magnetometer calibration, I always end up having most of the times negative Yaw value that changes accordingly when rotated but always returns to the same initial number. 

I am using STM32 and I changed the sample count of calibration from 1500 to 2000 with no results... 

I also set the magnetic declination for my location (which I translated into decimal because I wasn't sure if should use a value that was in degrees + minutes).

And I also changed the source code with OldSpice7's workaround to use ENU instead of NED (to fit my project needs).

Calibration most of the time seem to give similar bias values to these: -335.16, -36.55, -903.15
I literally tried to calibrate like 30 times and the results are always the same after I apply calibrated values... Yaw with at least -50 degree offset.

Thanks for any help in advance.

vollukas, 2021-04-02:
From what I was reading it's also not clear to me if the Yaw value should stay changed when I move it or if it should iterate back to the initial value like in my case... It seems weird to me that MPU 6050 calculates Yaw without having a magnetometer. I am clearly missing something and I'm not sure if I could just use gyro or acc rotation around Z-axis for my project if I just need a quadcopter to rotate in two directions at a given speed but without the need of a compass.

hideakitai, 2021-05-24:
@vollukas This issue will be fixed by https://github.com/hideakitai/MPU9250/tree/bugfix_obromious. Please confirm if it works.

hideakitai, 2021-05-28:
Fixed in v0.4.3. Please reopen the issue if you still have the problem.

closed 2021-05-28 by hideakitai