hideakitai/MPU9250 #45: Consistent use of fuse values

obromios, 2021-04-14:
This issue is based on a [discussion](https://github.com/kriswiner/MPU9250/issues/456#issue-836657973) in Kris Winer's repository.  This discussion has focused on whether there is a need to use the fuse values more consistently or if it is best to eliminate them entirely.  As per that discussion, I think that given all MPU9250 devices are over a year old, that it is best not to use the fuse values and so eliminate their use.  @hideakitai, what is your view on this.

hideakitai, 2021-05-24:
@obromios I've chosen to use your suggestion https://github.com/hideakitai/MPU9250/tree/bugfix_obromious. Please confirm if it works.

hideakitai, 2021-05-28:
Fixed in v0.4.3. Please reopen the issue if you still have the problem.

closed 2021-05-28 by hideakitai