hideakitai/MPU9250 #46: What is the correct way to transform the raw readings for the quaternion update

obromios, 2021-04-14:
There appears to be an inconsistency between this code base and the Kris Winer[ repository](https://github.com/kriswiner/MPU9250) in preparation for the quaternion update. In particular the code in this repository does the following transformation at line MPU9250.h:280
```
        float an = +a[0];
        float ae = -a[1];
        float ad = -a[2];
        float gn = +g[0] * DEG_TO_RAD;
        float ge = -g[1] * DEG_TO_RAD;
        float gd = -g[2] * DEG_TO_RAD;
        float mn = +m[1];
        float me = -m[0];
        float md = +m[2];
```
whereas the  Winer code at line MPU9250_MS5637_AHRS_t3.ino:521 does the following
`MadgwickQuaternionUpdate(-ax, ay, az, gx*PI/180.0f, -gy*PI/180.0f, -gz*PI/180.0f,  my,  -mx, mz);`

It can be seen that the gyro values are transformed in the same way but the accelerometer and magnetometer values are multipied by -1 in the Winer code compared to this code base.

Perhaps I have missed an earlier transformation or the quaternion filters use a different convention in the two code bases?

obromios, 2021-04-14:
@hideakitai, just to let you know this is the last issue I intend to raise before completing the pull request.

hideakitai, 2021-04-14:
@obromios This is because I thought I want to make the quaternion based on the real axes of IC, not based on the airplane axes. Though I have a mistake...

obromios, 2021-04-14:
So should I make them the same as the Winer code?

hideakitai, 2021-04-14:
No, please leave it as it is now.

obromios, 2021-04-15:
OK, I will close this thread. I have done a fork and started work on the changes for a pull request. Hopefully I will find time to finish it in a week or two.

closed 2021-04-15 by obromios