hideakitai/MPU9250 #47: Pitch not calculating correctly / Slow data

A-Basra, 2021-04-14:
Hi hideakitai,

Firstly, huge thanks! After spending far too long attempting to make my own library (as a novice programmer at best, yikes that was a shipwrek) your library is the only one which simply just works!

My first issue was with the wrap around effect on the roll which I found a solution to in another post though I will note that I have applied the updates to the MPU9250.h as specified in that previous issue. 

I'm having a couple slight issues. After calibration and once the data starts flowing my pitch access is returning inaccurate data. The data is 0 when in the starting position and then when tilted 90 degrees on its pitch axis the readings only go up to 35/40 degrees. The roll axis works fine and returns 90 (or there about) once tilted on the roll axis so the issue seems to be refined to the pitch. 

The second issue is that for both readings I seem to be getting a large spike in readings straight after setup finishes and the data from the loop starts flowing. See below data dump as reference. This was a test just running the example code and nothing else to ensure I wasn't causing any issues.

[code]
12:24:17.452 -> Accel Gyro calibration will start in 5sec.
12:24:17.452 -> Please leave the device still on the flat plane.
12:24:24.385 -> Mag calibration will start in 5sec.
12:24:24.385 -> Please Wave device in a figure eight until done.
12:24:29.442 -> Mag Factory Calibration Values: 
12:24:29.442 -> X-Axis sensitivity offset value 1.21
12:24:29.442 -> Y-Axis sensitivity offset value 1.21
12:24:29.442 -> Z-Axis sensitivity offset value 1.16
12:24:29.442 -> Mag Calibration: Wave device in a figure eight until done!
12:24:52.952 -> mag x min/max:
12:24:52.952 -> 148
12:24:52.952 -> 119
12:24:52.952 -> mag y min/max:
12:24:52.952 -> 364
12:24:52.952 -> 335
12:24:52.952 -> mag z min/max:
12:24:52.952 -> 328
12:24:52.952 -> 302
12:24:52.952 -> Mag Calibration done!
12:24:52.952 -> AK8963 mag biases (mG)
12:24:52.952 -> 240.70, 633.67, 549.80
12:24:52.952 -> AK8963 mag scale (mG)
12:24:52.952 -> 0.98, 0.98, 1.05
12:24:52.987 -> Mag Factory Calibration Values: 
12:24:52.987 -> X-Axis sensitivity offset value 1.21
12:24:52.987 -> Y-Axis sensitivity offset value 1.21
12:24:52.987 -> Z-Axis sensitivity offset value 1.16
12:24:52.987 -> < calibration parameters >
12:24:52.987 -> accel bias [g]: 
12:24:52.987 -> 923.56, -47.75, -961.08
12:24:53.022 -> gyro bias [deg/s]: 
12:24:53.022 -> 9.36, -4.65, -8.65
12:24:53.022 -> mag bias [mG]: 
12:24:53.022 -> 240.70, 633.67, 549.80
12:24:53.022 -> mag scale []: 
12:24:53.022 -> 0.98, 0.98, 1.05
12:24:53.022 -> Yaw, Pitch, Roll: -133.56, -9.45, -166.62
12:24:53.058 -> Yaw, Pitch, Roll: -133.72, -8.92, -167.47
12:24:53.095 -> Yaw, Pitch, Roll: -133.58, -8.50, -168.22
12:24:53.128 -> Yaw, Pitch, Roll: -133.80, -8.22, -168.96
12:24:53.128 -> Yaw, Pitch, Roll: -133.83, -7.76, -169.49
12:24:53.166 -> Yaw, Pitch, Roll: -134.16, -7.41, -170.06
12:24:53.200 -> Yaw, Pitch, Roll: -134.43, -7.18, -170.71
12:24:53.236 -> Yaw, Pitch, Roll: -134.50, -6.78, -171.17
12:24:53.236 -> Yaw, Pitch, Roll: -134.88, -6.34, -171.49
12:24:53.269 -> Yaw, Pitch, Roll: -134.84, -5.97, -171.99
12:24:53.306 -> Yaw, Pitch, Roll: -134.81, -5.61, -172.49
12:24:53.340 -> Yaw, Pitch, Roll: -134.92, -5.28, -173.01
12:24:53.373 -> Yaw, Pitch, Roll: -134.90, -5.11, -173.47
12:24:53.373 -> Yaw, Pitch, Roll: -134.77, -4.82, -173.83
12:24:53.408 -> Yaw, Pitch, Roll: -134.70, -4.56, -174.14
12:24:53.445 -> Yaw, Pitch, Roll: -134.77, -4.01, -174.29
12:24:53.478 -> Yaw, Pitch, Roll: -135.11, -3.82, -174.56
12:24:53.511 -> Yaw, Pitch, Roll: -135.33, -3.70, -174.88
12:24:53.511 -> Yaw, Pitch, Roll: -135.36, -3.62, -175.28
12:24:53.550 -> Yaw, Pitch, Roll: -135.31, -3.55, -175.51
12:24:53.587 -> Yaw, Pitch, Roll: -135.23, -3.44, -175.79
12:24:53.623 -> Yaw, Pitch, Roll: -135.75, -3.28, -175.89
12:24:53.623 -> Yaw, Pitch, Roll: -135.76, -3.19, -176.13
12:24:53.656 -> Yaw, Pitch, Roll: -135.81, -3.03, -176.32
12:24:53.689 -> Yaw, Pitch, Roll: -136.36, -2.97, -176.42
12:24:53.723 -> Yaw, Pitch, Roll: -136.32, -2.67, -176.48
12:24:53.756 -> Yaw, Pitch, Roll: -136.64, -2.47, -176.53
12:24:53.756 -> Yaw, Pitch, Roll: -137.02, -2.38, -176.66
12:24:53.790 -> Yaw, Pitch, Roll: -137.05, -2.40, -176.78
12:24:53.823 -> Yaw, Pitch, Roll: -137.35, -2.24, -176.93
12:24:53.857 -> Yaw, Pitch, Roll: -137.67, -2.20, -176.99
12:24:53.890 -> Yaw, Pitch, Roll: -138.04, -2.01, -176.95
12:24:53.923 -> Yaw, Pitch, Roll: -138.39, -1.75, -176.87
12:24:53.923 -> Yaw, Pitch, Roll: -138.33, -1.56, -177.04
12:24:53.957 -> Yaw, Pitch, Roll: -138.65, -1.42, -177.09
12:24:53.990 -> Yaw, Pitch, Roll: -138.52, -1.60, -176.99
12:24:54.023 -> Yaw, Pitch, Roll: -138.78, -1.64, -177.33
12:24:54.057 -> Yaw, Pitch, Roll: -138.93, -1.68, -177.38
12:24:54.057 -> Yaw, Pitch, Roll: -139.25, -1.74, -177.43
12:24:54.090 -> Yaw, Pitch, Roll: -138.89, -1.96, -177.48
12:24:54.125 -> Yaw, Pitch, Roll: -138.95, -2.11, -177.70
12:24:54.158 -> Yaw, Pitch, Roll: -138.95, -2.44, -178.09
12:24:54.191 -> Yaw, Pitch, Roll: -139.02, -2.41, -178.33
12:24:54.224 -> Yaw, Pitch, Roll: -139.37, -2.33, -178.48
12:24:54.224 -> Yaw, Pitch, Roll: -139.82, -2.23, -178.31
12:24:54.257 -> Yaw, Pitch, Roll: -140.09, -2.29, -178.34
12:24:54.293 -> Yaw, Pitch, Roll: -140.29, -2.45, -178.63
12:24:54.327 -> Yaw, Pitch, Roll: -140.60, -2.38, -178.54
12:24:54.360 -> Yaw, Pitch, Roll: -140.44, -2.40, -178.48
[/code]

As you can see there is a big fluctuation before the data levels out. I was wondering if there is a way to prevent this from happening? Presumably it's a result of the filters kicking in?

That brings me to my next issue I've been having which is speed. The data seems to be incredibly slow at responding to movement. Again I believe this may have something to do with the filters? I have read through the documentation and it's entirely possible I've seen the solution for this one and plainly just missed it from a lack of understanding but is there a way to bypass the filters and apply your own seperately? Diving into the source code I'm tempted to just go through and delete quat_filter and all references to it?

Thanks in advanced

hideakitai, 2021-05-24:
@A-Basra This issue will be fixed by https://github.com/hideakitai/MPU9250/tree/bugfix_obromious. Please confirm if it works.

hideakitai, 2021-05-28:
Fixed in v0.4.3. Please reopen the issue if you still have the problem.

closed 2021-05-28 by hideakitai