hideakitai/MPU9250 #51: quaternion output is broken

jarekdudzinski, 2021-05-14:
Hey. There is something very wrong with quaternion output...
When I set filter to NONE - all axes are correct, but there is strong drift when MPU just lies still.
Using MADGWICK or MAHONY - X and Z axes are reversed (in reference to accel & gyro axes printed on the MPU module).

Any solution to this?
I need the quaternion output, as most of 3D apps are using it (converted to rotation matrices), and also it doesn't generate gimbal lock problem.

jarekdudzinski, 2021-05-14:
PS: how I'm interpreting quaternions
1. sending values through serial to PC
2. on PC side, I'm using Processing script:
```
  Quaternion(float x, float y, float z, float w) {
    this.x = x; this.y = y; this.z = z; this.w = w;
  }
```
```
  PMatrix3D toMatrix(PMatrix3D out) {
    float x2 = x + x; float y2 = y + y; float z2 = z + z;
    float xsq2 = x * x2; float ysq2 = y * y2; float zsq2 = z * z2;
    float xy2 = x * y2; float xz2 = x * z2; float yz2 = y * z2;
    float wx2 = w * x2; float wy2 = w * y2; float wz2 = w * z2;
    out.set(
      1.0 - (ysq2 + zsq2), xy2 - wz2, xz2 + wy2, 0.0,
      xy2 + wz2, 1.0 - (xsq2 + zsq2), yz2 - wx2, 0.0,
      xz2 - wy2, yz2 + wx2, 1.0 - (xsq2 + ysq2), 0.0,
      0.0, 0.0, 0.0, 1.0);
    return out;
  }
```

hideakitai, 2021-05-24:
@jarekdudzinski This issue will be fixed by https://github.com/hideakitai/MPU9250/tree/bugfix_obromious. Please confirm if it works.

hideakitai, 2021-05-28:
Fixed in v0.4.3. Please reopen the issue if you still have the problem.

closed 2021-05-28 by hideakitai