hideakitai/MPU9250 #52: Is this a real MPU9250 chip ?

xXaRoXx, 2021-05-24:
I bought  MPU9250 modules off eBay and I think they aren't MPU9250s. They have the same i2c address (0x68) but they aren't recognized by this library. Can you help me identify the real chip please?  MP92 899LA1 1811 is printed on them and they have a "whoami" value of 112 (whatever "c" from the connection_check example means)
[image]

hideakitai, 2021-05-25:
Sorry, I don't know what it is. There are many imitations because MPU9250 has been discontinued a few years ago. Please ask the seller! I hope you can get one :)

closed 2021-05-25 by hideakitai

Hantuch, 2021-06-01:
Similiar case here, bought 1 exactly like that off on a renown local online seller for around $2 (50K IDR)

WHO_AM_I value was 78, with AK8963_WHOAMI at 0

Turns out i already got an older MPU9250 (with exact look), tested it, & it runs properly (whoami 71, ak8963 48)

I recommend report to the vendor and find another one 

[image]

rohit-kumar-j, 2021-06-14:
@xXaRoXx 
You can check this video (from 10:19) for the i2c call outs:
https://www.youtube.com/watch?v=mzwovYcozvI

hideakitai, 2021-08-28:
If AK8963 is just not present in that module, it may be usable by adding an additional AK8963 module. Unfortunately, if not, it's not available.

rohit-kumar-j, 2021-08-30:
@filipppp you could try [micro python ](https://micropython.org/)instead. There is pretty good documentation. However, you'll have to flash the chip with firmware using esptool. It is a little easier using python I guess.

Example Tutorial: https://www.youtube.com/watch?v=QopRAwUP5ds&ab_channel=ClaytonDarwin

MPU9250 in micropython: https://github.com/tuupola/micropython-mpu9250

paulosincos, 2021-09-30:
thanks @hideakitai .
I found another code that solved my problem.

I think will be nice if the your library print a more specific message to say the mpu6500 is not supported, or needs a external magnetometer.

paulosincos, 2021-09-30:
Another question:

If I remove the entire code of magnetometer (and related code), the other code will work with 6500? is it compatible?

paulosincos, 2021-09-30:
> Another question:
> 
> If I remove the entire code of magnetometer (and related code), the other code will work with 6500? is it compatible?

Answering myself, yes, I did it and it worked hehe

I just need to understand how to calculate Roll, Pitch and Yaw with just the gyroscope.

IanBUK, 2022-06-20:
> Sorry, I don't know what it is. There are many imitations because MPU9250 has been discontinued a few years ago. Please ask the seller! I hope you can get one :)

What's the recommended current 9dof imu please? A lot of reputable suppliers are still selling the mpu9250.

Thanks.

gree303, 2022-11-08:
> > Another question:
> > If I remove the entire code of magnetometer (and related code), the other code will work with 6500? is it compatible?
> 
> Answering myself, yes, I did it and it worked hehe
> 
> I just need to understand how to calculate Roll, Pitch and Yaw with just the gyroscope.

@paulosincos would you mind sharing your thoughts and workflow of swapping the mpu6050 vs the mpu9250 library.
I just received my esp32 + mpu9250 and ran into the same issues as you did. Any help is very welcome - thanks.

paulosincos, 2022-12-30:
> @paulosincos would you mind sharing your thoughts and workflow of swapping the mpu6050 vs the mpu9250 library. I just received my esp32 + mpu9250 and ran into the same issues as you did. Any help is very welcome - thanks.

@gree303 I'm working on a code implementation for fakes mpu9250.

I did this fork: https://github.com/paulosincos/MPU6500_aka-fake9250.

Unfortunately it's a partial solution to solve the problem.