hideakitai/MPU9250 #53: Different call functions

Tonymazzz, 2021-06-02:
Hi!  I was using other libraries from mpu9250 where I was using:

accelgyro.getMotion9(&ax, &ay, &az, &gx, &gy, &gz, &mx, &my, &mz);

But cannot call that with your library sir, its for my Air mouse, can you help me please?

Thanks in advance!!

Complete sketch:

// Watch video here: https://www.youtube.com/watch?v=o09-9n5WrxE

//Connection pins provided in the diagram at the beginning of video

// library provided and code based on: https://github.com/Seeed-Studio/IMU_10DOF
#include "Wire.h"
#include "I2Cdev.h"
#include "MPU9250.h"
#include <BleMouse.h>
#ifdef _ESP32_HAL_I2C_H_
#define SDA_PIN 21
#define SCL_PIN 22
#endif

BleMouse bleMouse;

// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for InvenSense evaluation board)
// AD0 high = 0x69
MPU9250 accelgyro;
I2Cdev   I2C_M;

int16_t ax, ay, az;
int16_t gx, gy, gz;
int16_t mx, my, mz;
int vx, vy;
float Gxyz[3];
int rightbutton = 16;
int leftbutton = 17;

void setup()
{
  // join I2C bus (I2Cdev library doesn't do this automatically)
  Wire.begin();
  Serial.begin(9600);
  Serial.println("Initializing I2C devices...");
  accelgyro.initialize();
  Serial.println("Testing device connections...");
  Serial.println(accelgyro.testConnection() ? "MPU9250 connection successful" : "MPU9250 connection failed");
  delay(1000);
  pinMode(leftbutton, OUTPUT);
  pinMode(rightbutton, OUTPUT);
  bleMouse.begin();
}

void loop()
{
  //getAccel_Data();
  getGyro_Data();

  float pitchrad = atan(Gxyz[0] / sqrt(Gxyz[1] * Gxyz[1] + Gxyz[2] * Gxyz[2])); // radians
  float rollrad = atan(Gxyz[1] / sqrt(Gxyz[0] * Gxyz[0] + Gxyz[2] * Gxyz[2])); // radians

  float rolldeg = 180 * (atan(Gxyz[1] / sqrt(Gxyz[0] * Gxyz[0] + Gxyz[2] * Gxyz[2]))) / PI; // degrees
  float pitchdeg = 180 * (atan(Gxyz[0] / sqrt(Gxyz[1] * Gxyz[1] + Gxyz[2] * Gxyz[2]))) / PI; // degrees
  
  /*float Min = -15;//-30, -45, -15
  float Max = 15;// 30, 45, 15
  int mapX = map(pitchdeg, Min, Max, -6, 6);
  int mapY = map(rolldeg, Min, Max, -6, 6);
  bleMouse.move(-mapX, mapY, 0);*/

  
 /* vx = (gx+100)/150;  // change 300 from 0 to 355
  vy = -(gz-100)/150; // same here about "-100"  from -355 to 0
   Mouse.move(vx, vy);*/
   
  Serial.print(pitchdeg);
  Serial.print(",");
  Serial.print(rolldeg);
  Serial.print(" - ");
  int leftstate = digitalRead(leftbutton);
  int rightstate = digitalRead(rightbutton);
  //Serial.print(leftstate); Serial.print(rightstate);
  if (leftstate == HIGH) {
    Serial.print(" Left Click! ");
    bleMouse.press(MOUSE_LEFT);
  }
  if (leftstate == LOW) {
    bleMouse.release(MOUSE_LEFT);
  }
  if (rightstate == HIGH) {
    Serial.print(" Right Click! ");
    bleMouse.click(MOUSE_RIGHT);
  }
  Serial.println();
}

//void getAccel_Datavoid
//{

 /* accelgyro.getMotion9(&ax, &ay, &az, &gx, &gy, &gz, &mx, &my, &mz);
  Axyz[0] = ((double) ax / 256) - 1.6;
  Axyz[1] = ((double) ay / 256) - 2.1;
  Axyz[2] = (double) az / 256;*/
/*SI CAMBIO ESTO POR
 *     Axyz[0] = (double) ax / 16384;
    Axyz[1] = (double) ay / 16384;
    Axyz[2] = (double) az / 16384;   
    
    EL RATON SE DESVIA MAS RAPIDO QUE CON EL DE ARRIBA
    Y SI LO QUITO NO FUNCIONA*/

    void getGyro_Data(void)
{
    accelgyro.getMotion9(&ax, &ay, &az, &gx, &gy, &gz, &mx, &my, &mz);
    Gxyz[0] = (double) gx * 250 / 32768;
    Gxyz[1] = (double) gy * 250 / 32768;
    Gxyz[2] = (double) gz * 250 / 32768;
//}
 
    vx = (gx+100)/150;  // change 300 from 0 to 355
  vy = -(gz-100)/150; // same here about "-100"  from -355 to 0

     bleMouse.move(vx, vy);
}

hideakitai, 2021-06-03:
Hi, please read README carefully and you can find how to use this library. The examples will also help you to understand this library.

Tonymazzz, 2021-06-03:
Hi sir, 

Thanks for your quick response and clarification.

I've been reading all day and testing your examples but I can't get the mouse to move yet, could you help me please dear Hideakitai?

Thanks in advance master!!!

hideakitai, 2021-06-04:
I can only help you with errors and usage that are not covered in the README or examples of this library. Please note that I cannot help you with your algorithm or implementation itself. Please tell me exactly what the problem is with this library.

Tonymazzz, 2021-06-04:
Hi sir,

I just need to know the alternative of use:
accelgyro.getMotion9 (& ax, & ay, & az, & gx, & gy, & gz, & mx, & my, & mz);

Coz accelgyro.getMotion9 dosnt t work with you library dear Hideakitai.

 Any idea sir?

hideakitai, 2021-06-04:
@Tonymazzz You can get the sensor data by following methods. Please read the README.

``` C++
float getAccX() const;
float getAccY() const;
float getAccZ() const;
float getGyroX() const;
float getGyroY() const;
float getGyroZ() const;
float getMagX() const;
float getMagY() const;
float getMagZ() const;
```

Tonymazzz, 2021-06-05:
Thanks for the aclaration sir!!!!

But im Lost with the next so...

vx = (gx+100)/150; // change 300 from 0 to 355
vy = -(gz-100)/150; // same here about "-100" from -355 to 0

Its not working :$

hideakitai, 2021-06-06:
@Tonymazzz Sorry, I can't help debugging of your own project. If you want me to write and debug the code instead of you, let me know from email as a job, as mentioned in #40.

closed 2021-06-06 by hideakitai