hideakitai/MPU9250 #54: SoftWire + MPU9250

longo92-vrmedia, 2021-06-16:
Hi,
i''m using the library with arduino nano v3.0 and the software i2c on D2 (SDA) and D3(SCL). Using the basic example, when querying WHO_AM_I_MPU9250 i get 0xFF (instead of 0x71, in the setup function). Can you help me?
Here is the code:
#include "MPU9250.h"
#include "SoftWire.h"
#include <AsyncDelay.h>

MPU9250_<SoftWire> mpu;
//MPU9250 mpu;

SoftWire sw(2, 3);
char swTxBuffer[32] = {0}; 
char swRxBuffer[32] = {0};

// you need setting struct
MPU9250Setting setting;

void setup() {
    Serial.begin(115200);
    delay(2000);
    sw.setTxBuffer(swTxBuffer, sizeof(swTxBuffer));
    sw.setRxBuffer(swRxBuffer, sizeof(swRxBuffer));
    sw.setClock(400000);
    sw.begin();

    if (!mpu.setup(0x68, setting, sw)) {  // change to your own address
        while (1) {
            Serial.println("MPU connection failed. Please check your connection with `connection_check` example.");
            delay(5000);
        }
    }
}

void loop() {
    if (mpu.update()) {
        static uint32_t prev_ms = millis();
        if (millis() > prev_ms + 25) {
            print_roll_pitch_yaw();
            prev_ms = millis();
        }
    }
}

void print_roll_pitch_yaw() {
    Serial.print("Yaw, Pitch, Roll: ");
    Serial.print(mpu.getYaw(), 2);
    Serial.print(", ");
    Serial.print(mpu.getPitch(), 2);
    Serial.print(", ");
    Serial.println(mpu.getRoll(), 2);
}

hideakitai, 2021-06-18:
@longo92-vrmedia Sorry for the late reply. Could you try this line as shown in README?

```c++
MPU9250_<SoftWire, MPU9250_WHOAMI_DEFAULT_VALUE> mpu;
```

longo92-vrmedia, 2021-06-18:
@hideakitai  I've hust tried, but  it gives me:

MPU9250_SIMPLE:6:47: error: wrong number of template arguments (2, should be 1)
 MPU9250_<SoftWire,MPU9250_WHOAMI_DEFAULT_VALUE> mpu;

And digging in your code only the WireType is used in template
                                               ^

hideakitai, 2021-06-18:
@longo92-vrmedia Sorry, you don't need second argument

```c++
MPU9250_<SoftWire> mpu;
```

longo92-vrmedia, 2021-06-18:
Sorry, i posted the example wrong code, but in my trials i always use  MPU9250<SoftWire> mpu.
Try this code:
#include "MPU9250.h"
#include "SoftWire.h"
#include <AsyncDelay.h>

MPU9250_<SoftWire> mpu;
//MPU9250 mpu;

SoftWire sw(2, 3);
char swTxBuffer[32] = {0}; 
char swRxBuffer[32] = {0};

// you need setting struct
MPU9250Setting setting;

void setup() {
    Serial.begin(115200);
    delay(2000);
    sw.setTxBuffer(swTxBuffer, sizeof(swTxBuffer));
    sw.setRxBuffer(swRxBuffer, sizeof(swRxBuffer));
    sw.setClock(400000);
    sw.begin();

    if (!mpu.setup(0x68, setting, sw)) {  // change to your own address
        while (1) {
            Serial.println("MPU connection failed. Please check your connection with `connection_check` example.");
            delay(5000);
        }
    }
}

void loop() {
    if (mpu.update()) {
        static uint32_t prev_ms = millis();
        if (millis() > prev_ms + 25) {
            print_roll_pitch_yaw();
            prev_ms = millis();
        }
    }
}

void print_roll_pitch_yaw() {
    Serial.print("Yaw, Pitch, Roll: ");
    Serial.print(mpu.getYaw(), 2);
    Serial.print(", ");
    Serial.print(mpu.getPitch(), 2);
    Serial.print(", ");
    Serial.println(mpu.getRoll(), 2);
}

As said, still gettin 0xFF in line 180 of MPU9250.h :
  bool isConnectedMPU9250() {
        byte c = read_byte(mpu_i2c_addr, WHO_AM_I_MPU9250);
        if (b_verbose) {
            Serial.print("MPU9250 WHO AM I = ");
            Serial.println(c, HEX);
        }
        bool b = (c == MPU9250_WHOAMI_DEFAULT_VALUE);
        b |= (c == MPU9255_WHOAMI_DEFAULT_VALUE);
        b |= (c == MPU6500_WHOAMI_DEFAULT_VALUE);
        return b;
    }

hideakitai, 2021-06-18:
@longo92-vrmedia Can you try `connection_check` example with SoftWIre first?

longo92-vrmedia, 2021-06-18:
Yes, this is the output:
Searching for i2c devices...
Found 1 I2C devices
I2C addresses are: 0x68 
I2C address 0x68 is not MPU series. Please use correct device
The error si the same:
` byte ca = readByte(addrs[i], WHO_AM_I_MPU9250,sw);`
Here it returns 0xFF

hideakitai, 2021-06-18:
Hm, can you read the WHO_AM_I register directly with SoftWire?

hideakitai, 2021-06-23:
@longo92-vrmedia I found that the `SoftWire` library fails to read the data. This is the transaction with `Wire` library:

[image]

and this is the transaction of `SoftWIre` library with same setting:

[image]

So `requestFrom` function fails to send read command. I will continue to explore what is the problem with `SoftWire` library.

hideakitai, 2021-06-24:
@longo92-vrmedia I've added `connection_check_softwire` example in v0.4.5. Maybe this example will work, but main functions of this library will NOT work with current `SoftWire` as is. The problem is repeated start is not issued by `SoftWire` (see stevemarple/SoftWire#22), so if you want to use this library with `SoftWire`, please try to remove `false` from these lines in `MPU9250.h`:

https://github.com/hideakitai/MPU9250/blob/98ca851dba3376e3171b18a9914eece129d16576/MPU9250.h#L953

https://github.com/hideakitai/MPU9250/blob/98ca851dba3376e3171b18a9914eece129d16576/MPU9250.h#L963

hideakitai, 2021-08-02:
@longo92-vrmedia Closing this issue because this is not the problem of MPU9250 library, but SoftWire. Thank you for reporting.

closed 2021-08-02 by hideakitai