hideakitai/MPU9250 #55: Which eeprom library is compatible with the this?

adam515, 2021-06-28:
Hi, thank you for creating such a great library!
I'd like to use the saveCalibration() function on Teensy 4.1, but could not find a compatible 
#include "eeprom_utils.h" to use.

Could you please clarify?

hideakitai, 2021-06-28:
`eeprom_utils.h` should be in the same folder as `calibration_eeprom.ino.` If you use the `calibration_eeprom` folder as is, it should be compiled. The folder looks like [this](https://github.com/hideakitai/MPU9250/tree/master/examples/calibration_eeprom).

adam515, 2021-06-28:
thank you, got it!

I don't see in your library support for tilt-compensated yaw measurement (absolute heading) that always give a const heading irrespective of pitch and roll. Would you have any pointers on that?

hideakitai, 2021-06-28:
What is the difference between roll, pitch, and yaw, which are supported by this library? You can change magnetic declination.

adam515, 2021-06-28:
With the current setup, calling getYaw() returns different values as pitch and roll change. I'd like to remove those effects and only measure heading that doesn't change by tilting the MPU9250. Only if the MPU is rotated then this heading would change. Do you recommend a clean way to do it?

hideakitai, 2021-06-28:
I think it's because a calibration is not sufficient. Please re-calibrate your device.

hideakitai, 2021-07-02:
If you still have an issue with this library, please let me know.

closed 2021-07-02 by hideakitai