hideakitai/MPU9250 #56: Question: mahony filter - how to ensure it runs 10-20 times before new readings?

papparazzi, 2021-08-02:
Hello, 

I adopted your library for a VR sensor project. What I understand from reading through Kris Winer's github and other sources is that the Filters should best run 10 to 20 iterations before new data is added. 

I have not found out how to implement this with your library out of the box and therefore ended up introducing a do-while loop in update() to execute the filter 20 times. 

```
bool update() {
if (available())
        {
        update_accel_gyro();
        update_mag();
       }
int i = 1 ;
            do{    
               quat_filter.update(+a[0], +a[1], +a[2], +g[0] * DEG_TO_RAD, +g[1] * DEG_TO_RAD, +g[2] * DEG_TO_RAD, +m[1], +m[0], -m[2], q);
            i++;    
            } while (i <= 20) ;
}
```
Is there any other / better way of achieving this ? 

Hardware:
- ESP32 Wrover-B
- MPU9250 breakout board 

Thanks

hideakitai, 2021-08-02:
Hi, there is no mechanism in this library to apply the same value ​​to the filter over and over again. 

Does “iteration” mean applying the same value to a filter over and over again?  I wondered that it means 10-20 iterations are required to converge to the correct value. If you have an information source, please let me know.

papparazzi, 2021-08-02:
Hi, I have read this multiple times in the Git issue discussions of Kris Winer - for example here https://github.com/kriswiner/MPU9250/issues/420. or https://github.com/kriswiner/MPU9250/issues/427

His code examples are not making this very explicit, he explained in some thread that by having a lower FIFO frequency  the fusion filter automatically get's run multiple times (at least 5 times is apparently desirable) as there would be no interrupt and therefore no new readings wherefore the loop() essentially skips reading sensor value but instead runs the fusion filter one (mutliple) times with the old variables again. 

I believe my simplified code example above works but there might be a better way of doing it with your library.

hideakitai, 2021-08-03:
Oh really, I didn't know that. I'll add it as an option! But the default value is only one iteration. Because I think it's not so unstable as is.

hideakitai, 2021-08-03:
Added this feature in v0.4.6. Closing.

```C++
mpu.setFilterIterations(10);
```

closed 2021-08-03 by hideakitai