hideakitai/MPU9250 #62: Yaw Values

beratgnr, 2021-09-24:
Yaw values does not match with compass. Also when I rotate the sensor 90 degree , Yaw value changes 40-50 degree. I can not fix that , I am using simple with callibration code. How can I fix that.

hideakitai, 2021-09-25:
Hi, there seems to be a problem with the calibration. Can you try again?

beratgnr, 2021-09-25:
Hi, thank you for answering. Do I need calibrate the magnetometer in flat plane or 3D calibration with rotating every angle?

hideakitai, 2021-09-30:
First, please stay still on the flat plane until the accel/gyro calibration finishes. After that, rotate the device around.

beratgnr, 2021-09-30:
Thank you

30 Eyl 2021 Per 14:19 tarihinde Hideaki Tai ***@***.***> şunu
yazdı:

closed 2021-09-30 by hideakitai

MotoMotoRo, 2022-05-08:
Hello !
how can I change the magnetometer scale from 180 >> 0 << - 180 to 0 >> 360 ...?
Thank you in advance.

beratgnr, 2022-05-31:
> Hello ! how can I change the magnetometer scale to 0 >> 360 ...? Thank you in advance.
Maybe belove algorithm can work. 

[image]

MotoMotoRo, 2022-05-31:
Thanks for the help.