hideakitai/MPU9250 #65: mpu.setup call feezing

paulosincos, 2021-09-27:
Hello.

I'm triyng to use you library, but my sketch is freezing when I call mpu.setup.

your connection check sketch log:

```
13:15:47.935 -> Searching for i2c devices...
13:15:47.935 -> Found 1 I2C devices
13:15:47.935 -> I2C addresses are: 0x68 
13:15:47.935 -> I2C address 0x68 is MPU6500 and ready to use
```

my sketch:

```c
#include "MPU9250.h"

MPU9250 mpu;

void setup() {
    Serial.begin(115200);
    Wire.begin();
    delay(2000);
    Serial.println("Starting MPU");
    mpu.setup(0x68);  // <---------- freezes here!!!
    Serial.println("MPU Started");
}

void loop() {
    if (mpu.update()) {
        Serial.print(mpu.getYaw()); Serial.print(", ");
        Serial.print(mpu.getPitch()); Serial.print(", ");
        Serial.println(mpu.getRoll());
    }
}
```

and my log is only:

```
13:16:11.632 -> Starting MPU
```

I'm using an Arduino Nano and an MPU described as "mpu-9250/6500/9255".

I checked the weldering and all looks fine.

paulosincos, 2021-09-27:
I'm using the last release version, 0.4.6

paulosincos, 2021-09-27:
[image]

my connections

hideakitai, 2021-09-30:
@paulosincos Hi, your device is not MPU9250 but MPU6500. MPU6500 doesn't have a magnetometer, so you can't use it with this library. Please also refer #52

closed 2021-09-30 by hideakitai