hideakitai/MPU9250 #70: Help! How to convert quaternion from right-handed to left-handed?

qazujmko, 2021-10-30:
I get [w,x,y,z] from your api.  (Right-Handed, X-forward, Z-down) 
I want to convert it to left-handed,X-right,Z-backward.
I dont know math. Please tell me the result. It's driving me mad.

qazujmko, 2021-10-31:
I think I get something wrong. Anyway I want to get a X-right,Y-up,Z-backward result.

hideakitai, 2021-11-01:
You can use the tool called "Google" to search for a solution. e.g., https://stackoverflow.com/questions/28673777/convert-quaternion-from-right-handed-to-left-handed-coordinate-system

hideakitai, 2021-11-01:
Maybe this one is a similar situation with you https://gamedev.stackexchange.com/questions/157946/converting-a-quaternion-in-a-right-to-left-handed-coordinate-system

closed 2021-11-01 by hideakitai