hideakitai/MPU9250 #72: Magnetometer always zero, Accel + Gyro working fine

darrenrahnemoon, 2021-11-14:
Hi guys,

I have the following code: 
```
#include <MPU9250.h>
#include <Wire.h>

MPU9250 sensor;

void setup() {
	Serial.begin(115200);
	Serial.println("Setup Started.");
	Wire.begin();

	delay(2000);
	setupSensor();

	Serial.println("Setup Finished.");
	delay(1000);
}

void loop() {
	if (sensor.update()) {
		clearScreen();
		Serial.println("Roll: " + (String)sensor.getRoll() + ", Pitch: " + (String)sensor.getPitch() + ", Yaw: " + (String)sensor.getYaw());
		Serial.println("Accelerometer X: " + (String)sensor.getAccX() + ", Y: " + (String)sensor.getAccY() + ", Z: " + (String)sensor.getAccZ() + " (m/s^2)");
		Serial.println("Gyroscope X: " + (String)sensor.getGyroX() + ", Y: " + (String)sensor.getGyroY() + ", Z: " + (String)sensor.getGyroZ() + " (rads/s)");
		Serial.println("Compass X: " + (String)sensor.getMagX() + ", Y: " + (String)sensor.getMagY() + ", Z: " + (String)sensor.getMagZ() + " (uT)");
		Serial.println("Temperature: " + (String)sensor.getTemperature() + " ˚C");
	}
	delay(100);
}

// Helper functions
void clearScreen() {
	Serial.write(27);		// ESC command
	Serial.print("[2J");	// clear screen command
	Serial.write(27);
	Serial.print("[H");	 // cursor to home command
}

void setupSensor() {
	sensor.setup(0x68);
	delay(5000);
	calibrateSensor();
}

void calibrateSensor() {
	Serial.println("Calibrating Accelerometer and Gyroscope.");
	sensor.calibrateAccelGyro();

	Serial.println("Calibrating Compass.");
	sensor.calibrateMag();
}
```

When I run this I do get some values for the accelerometer and gyroscope. However my magnetometer doesn't return anything (always set to zero)

Sample:

Chrisrdouglas, 2021-11-28:
I had that issue with one of the ones i ordered. The device I ordered was defective. ordered a new one and ran my same code a few days later and it worked

A7-00, 2021-12-21:
> I had that issue with the magnetometer on one of the ones I ordered. The device I ordered was defective. ordered a new one and ran my same code a few days later and it worked

I will have to disagree with you on that @Chrisrdouglas , I managed to write a code that works consistently well, and another that has that problem on the exact same setup, I have no idea why that really happens but it seems to be a timing issue. I am curious tho to see if the working code works constantly on multiple platforms as I'm using the esp8266 rather than the Arduino nano so @darrenrahnemoon I would like you to try it and see if it does work for you and tell us about it.

working code

#include <Arduino.h>
#include <MPU9250.h>

#define I2CADDRESS 0x68

MPU9250 mpu;

float ypr[3];

void setup() {

  Serial.begin(115200);
  Wire.begin();
  if (!mpu.setup(I2CADDRESS)) {
    while (1) {
      Serial.println("MPU connection failed. Please check your connection with `connection_check` example.");
      digitalWrite(LED_BUILTIN, LOW);
      delay(100);
      digitalWrite(LED_BUILTIN, HIGH);
      delay(100);
    }

  }

  Serial.println("Accel Gyro calibration will start in a sec.");
  Serial.println("Please leave the device still on the flat plane.");
  mpu.verbose(true);
  delay(1000);
  mpu.calibrateAccelGyro();
  Serial.println("Mag calibration will start in a sec.");
  Serial.println("Please Wave device in a figure eight five times.");
  delay(1000);
  mpu.calibrateMag();
  mpu.verbose(false);

}

void loop() {

  if (mpu.update()) {
    ypr[0] = mpu.getEulerZ();
    ypr[1] = mpu.getEulerY();
    ypr[2] = mpu.getEulerX();
    Serial.printf("yaw: %f, pitch: %f, roll: %f\n", ypr[0], ypr[1], ypr[2]); 
  }

}

Chrisrdouglas, 2021-12-21:
Oh, that's interesting! So if it was a timing issue then would that only affect the values for the magnetometer? Or would the values for the gyro and accelerometer be affected as well?

A7-00, 2021-12-21:
@Chrisrdouglas that is a great point.
I have no clue why only the magnetometer value is the only one that gets affected, but I think I pinned it down the issue happens if I delay mpu.update() with more than 15mSec, no clear idea why and if it is only on some version of the sensor or not. It is different than what I thought initially I thought it had to be something with the calibration but no it's not. I also would love to know if you experience the same issue @Chrisrdouglas on your setup.

Chrisrdouglas, 2021-12-21:
So when I ran the calibration code it was on a Teensy 3.2 and I used the calibration.ino file. I didn't do anything fancy with it at that point. That's when I discovered that the magnetometer was just reading all zeros. The second one I got worked flawlessly. I think that's all the information I have for ya. 

Here's the one I purchased: http://www.hiletgo.com/ProductDetail/1953399.html

hideakitai, 2021-12-22:
Maybe related to #52

hideakitai, 2022-01-02:
I am closing this is not a problem for the library (the device is not real MPU9250).

closed 2022-01-02 by hideakitai

Nahida-Majeed, 2023-01-19:
hi how to check, the value (x,y,z) from the magnetometer sensor is correct reading or not?

Chrisrdouglas, 2023-01-22:
@Nahida-Majeed,

Yse `your_mpu.getMagX()`, `your_mpu.getMagY()`, and `your_mpu.getMagZ()`. if you're getting zeros then you most likely got sold a defective/fake copy.

I'd verify the output by moving a magnet near one of the sides and seeing how it affects the value. if it goes up while the magnet gets closer then i would assume that means you got a good one