hideakitai/MPU9250 #75: read gyro value when max30102 sensor is also connected

karakuve, 2022-01-25:
I think this is a bug ,
when i have both sensor connected either in the same i2c port or different i2c on the same mcu, the gyro does not return any value, it only display 0 for all 3 value which in my case i use yaw, roll, pitch. I already try for euler value, also the same.

so in void loop, i usually collecting a few data for the max30102 then i read the data from mpu9250. what i do to avoid return 0 value. i make sure in void loop the first thing i do is to read mpu9250 a few times consecutively , then i proceed with the usual flow that i have. and it fix the problems.

however , the value from the mpu9250 fluctuate a little bit which not happening when only mpu9250 is connected. i not yet play around with the mpu9250 setting. just want to let you know, i think it is a bug but im not sure.

im using:
esp32 dev board - wrover e
platform io v2.4.0
arduino framework
mpu9250 library v 0.4.8
sparkfun mpu9250 - v1.1.1

hideakitai, 2022-01-25:
I think it's not a library bug if it works without any other libraries. Could you try to analyze your circuit and let me know about the detail if you still have the problem?

karakuve, 2022-01-25:
Sure i will have a look again, i not yet analyse it with logic analyser, but what i can say using mpu9250 with mlx90614 based on adafruit library ,does not produce this kind of problems.

hideakitai, 2022-03-07:
I am closing this issue because no further information has been provided. Please feel free to reopen this.

closed 2022-03-07 by hideakitai