hideakitai/MPU9250 #77: Need list of changes from the original library.

adityasivaraj, 2022-02-10:
Hi, this is great work. I would like to port a program written for the original Kris Winer library, to your API. Do you have a list of changes you made to the original library, like function names modified, functions added/removed, changes in access modifiers etc.,?

hideakitai, 2022-02-10:
Hi, Kris's repository I referred to is a code snippet, not a library. Which library are you using?

adityasivaraj, 2022-02-10:
I'm sorry, I meant to write Kris Winer's code. I got the impression that you have converted his code into a library (MPU9250.h). He also seems to have quaternionFilters as another Arduino file that needs to be kept in the same directory. So I thought you ported his code as a standard Arduino MPU9250 library.

hideakitai, 2022-03-09:
@adityasivaraj Sorry for the late reply. Yes, I've re-organized his code as a standard Arduino library. I've divided his code into several functions first. But currently there are many original features, I can't compare this library with his code simply. If you have still have the question, please feel free to add comments.

closed 2022-03-09 by hideakitai