hideakitai/MPU9250 #80: Roll and Pitch affect the Yaw

kyrvayapierdole, 2022-02-16:
After calibration i got heading very near that my iphone says. But when i pitch it`s effect very strongly on the Yaw. Also one strange thing that when heading 180 (South) it doe`s not effect but in other scenarios it does.

hideakitai, 2022-02-24:
Gimbal lock? Please refer https://github.com/hideakitai/MPU9250/issues/69#issuecomment-948356416

abhijitmajumdar, 2022-02-28:
Not a solution, but this worked for me (with a tradeoff), so maybe this works with your use case.

I had the same issue when using this library. Following through the comments here and https://github.com/hideakitai/MPU9250/issues/69#issuecomment-948356416, specifically as @hideakitai mentions,

> I don't know if this is truly MPU-9250, but I can't support it if it isn't (not compatible with MPU-9250)

I realized I was indeed interfacing with a MPU-9265 (not sure of the differences, maybe a knockoff?). That said, I was able to resolve my issue by simply switching the filter to `MAHONY` (following instructions from https://github.com/hideakitai/MPU9250#quaternion-filter)

```
MPU9250 mpu;

void setup() {
  ...
  if (!mpu.setup(0x68)){
    Serial.println("Could not initialize MPU");
  }
  else{
    mpu.selectFilter(QuatFilterSel::MAHONY);
  }
}
```

Caveat: Switching to the `MAHONY` filter introduced observable drift (~0.8deg/s) in yaw (not so much in roll nor pitch). This might probably be because of bad calibration, but reverting back to the `MADGWICK` filter did not drift as much

kyrvayapierdole, 2022-03-04:
@abhijitmajumdar @hideakitai Thanks for yours answers. Unfortunately because of war (I'm from Ukraine) i can't now test it. But i hope very soon i will test everything and give some information and researchings.

hideakitai, 2022-03-05:
May peace be upon us all soon!

kyrvayapierdole, 2022-04-16:
Hello, sorry for late reply. @abhijitmajumdar your trick works but also with yaw draft. @hideakitai Today after about 30 different calibrations i found that everything works fine when i give roll 90 deg or more. It's looks like gimbal lock issue. So in position that i need it not compensate yaw with roll and pitch. But when I flip it upside down it works just fine. The problem is I don;t understand how can I calibrate it properly with out this lock. I was looking at yours link about gimbal lock and other resources and there is nothing about solving it. Any ideas?

kyrvayapierdole, 2022-04-30:
Hello. I resolve my problem. The reason is that calibration is not enough. But problem is there is small amount information about proper calibration. I found that [tutorial](https://www.instructables.com/Easy-hard-and-soft-iron-magnetometer-calibration/) and it worked! Paper based on HMC5883L magnetometer module, but it's not problem, all you and pass row data to serial. There is a tool inside this tutorial calls MagViewer. You cann just pass you raw mag data and check calibration. I saw that my sphere are not sphere at all and also it's a bit not in center. That was the reason why i'm getting this affect from roll and pitch. So before calibration I set filter to NONE

kyrvayapierdole, 2022-04-30:
Hello. I resolve my problem. The reason is that calibration is not enough. But problem is there is a small amount information about proper calibration. I found that [tutorial](https://www.instructables.com/Easy-hard-and-soft-iron-magnetometer-calibration/) and it worked! Paper based on HMC5883L magnetometer module, but it's not a problem. All you need pass row mag data to serial. There is a tool inside this tutorial calls MagViewer. You can just pass you raw mag data and check calibration. I saw that my sphere are not sphere at all and also it's a bit not in center. That was the reason why i'm getting this affect from roll and pitch. 

So before calibration I set filter to NONE and set my Declination that it.

```
    mpu.selectFilter(QuatFilterSel::NONE);
    mpu.setMagneticDeclination(+8.28);
```

After it you only need to log your mag data to serial with 9600 rate.

```
    Serial.print(mpu.getMagX());
    Serial.print(",");

    Serial.print(mpu.getMagY());
    Serial.print(",");

    Serial.print(mpu.getMagZ());
    Serial.println(",");
```

Do every thing like paper says and you will get your matrix and bias. Mine was like this.

[image]

Bias it's simple bias, you set it like this

`mpu.setMagBias(472.602, 512.183, -836.117);`

And your matrix. I don't know maybe it's bad conversion and that is not correct but it worked for me.

I add values 
M11 + M12 + M13 = ScaleX
M21 + M22 + M23 = ScaleY
M31 + M32 + M33 = ScaleZ

In my case it was 2.523, 2.301, 2.22. And it's was perfectly fine. But all calibration (build-in and others) return scale like 1.f not 2.f so I divide my values by 2 and got 1.2615, 1.1505, 1.11. And it's seems works the same.

After all check inside MagViewer and I see almost perfect sphear. And it was very small offset on z axis. So i manually edit the value just subtract like 30 and now sphear at (0;0).

After all enable MADGWICK filter and it's works like expected. Tilt affect about 1-2 degress for 180 angle tilt. I think if manually edit few values it will be like perfect.

I don't know why but built-in calibrations doesn't works for me properly. Actually maybe I do something wrong, as I said there is small amount  information about it. My problem resolved and this issue closed. If anyone having any problem with calibration, fell free to ask.

closed 2022-04-30 by kyrvayapierdole