hideakitai/MPU9250 #82: Questions about the 25 ms delay

hsiachi, 2022-04-22:
Hi, thank you for maintaining this project!
I'm using the library with its calibration, but I notice that if I comment out the 25 ms delay, quaternions will not be updated (More specifically it's `1, 0, 0, 0`).
```arduino
    if (mpu.update()) {
        static uint32_t prev_ms = millis();
        if (millis() > prev_ms + 25) {
            print_roll_pitch_yaw();
            prev_ms = millis();
        }
    }
```
I've read the source code and could not figure out why it's happening, since everything seems to be synchronous. But maybe I am not careful enough and miss something important. It will be very kind of you if you could help me out!

hideakitai, 2022-04-22:
Hi, could you fill the issue template, please?

hsiachi, 2022-04-23:
> Hi, could you fill the issue template, please?

I've updated with the issue template now :)

hideakitai, 2022-04-24:
Could you give me "Reproducible & Minimal & Compilable Code" if you modified my example? I don't know what `print_quaternion` is.

hsiachi, 2022-04-24:
Certainly. Sorry for bothering you

hsiachi, 2022-04-30:
I figured out that it's because I've changed the baud rate to 9600 bits/sec (which I didn't notice). The sensor runs at 200 Hz, and printing a quaternion just exceeds the limit (while printing Euler Angles does not). Changing to 115200 bits/sec solves my problem. Sorry again for taking your time.

closed 2022-04-30 by hsiachi