hideakitai/MPU9250 #83: Read sensor values only, without estimating attitude

aryadas98, 2022-05-08:
Thank you for the excellent library. We are working on a research project where we would like to use custom attitude estimation algorithms. In this library calling the `update()` function calls the corresponding filter routine as well. This would result in unnecessary computation. We would just like to read the acc, gyro and mag values in proper units without any other processing.

kyrvayapierdole, 2022-05-09:
@aryadas98 Hey. You can set the filter to none like this:

`mpu.selectFilter(QuatFilterSel::NONE);`

You want to do it after mpu.setup inside setup function but ofcourse you can do it anywhere you want.

Then you just read data with this functions:

```
float getAccX() const;
float getAccY() const;
float getAccZ() const;
float getGyroX() const;
float getGyroY() const;
float getGyroZ() const;
float getMagX() const;
float getMagY() const;
float getMagZ() const;
```

You can see all functions by this [link](https://github.com/hideakitai/MPU9250#apis)

aryadas98, 2022-05-11:
But even then, the gyro integration calculation is done. For our purposes, we decided to modify the `update()` function as follows:

```
bool update() {
        if (!available()) return false;

        update_accel_gyro();
        update_mag();

        return true;
    }
```

This avoids any unnecessary calculation. Just reads the data.

closed 2022-05-11 by aryadas98