hideakitai/MPU9250 #88: Accelerometer Calibration isn't Implemented, effects Roll / Pitch / Yaw Data

RobertJN64, 2022-09-07:
## Environment

IDE: PlatformIO
Board/MCU: ESP32 Devkit
MPU-9250 Library Version: 0.4.8

It looks like the accelerometer calibration code was removed from the library. I added this code to the `update_accel_gyro()` function, and got much better results:

```c
a[0] -= acc_bias[0] * 2 / 32768.f; //handle calibration
a[1] -= acc_bias[1] * 2 / 32768.f;
a[2] -= acc_bias[2] * 2 / 32768.f;
```

I would submit a PR, but I haven't properly tested for different resolutions, and I don't know if you want to try to reimplement the code that saves this calibration to the actual device.

Woromi, 2023-03-19:
This shouldn't be necessary, since MPU9250 itself adds the value of bias registers to the measured value. For some reason, this doesn't happen after calibration. I haven't figured out why, but reading the bias registers (getAccBias) and setting them again (setAccBias) manually solves it.

Berg0162, 2023-04-01:
Came across the same problem... Please check the following simple change in the code that results in an improved outcome, comparable with [RobertJN64](https://github.com/RobertJN64) suggestion (see above)...!
```C++
    void calibrate_acc_gyro_impl() {
        set_acc_gyro_to_calibration();
        collect_acc_gyro_data_to(acc_bias, gyro_bias);
        // New acc_gyro biasses are collected during calibration, we want to apply these during further operation
/* original code sequence:
        write_accel_offset();
        write_gyro_offset();
        delay(100);
        initMPU9250();
        delay(1000);
*/
// --> start the sequence with initializing MPU9250 and only afterwards set the values of the hardware registers... 
        initMPU9250();        // Toggle Reset Device --> clears ALL(!) registers --> also acc_gyro bias registers
        delay(1000);          // stabilize
        write_accel_offset(); // Push accelerometer biases to hardware registers
        write_gyro_offset();  // Push gyrometer biases to hardware registers
        delay(100);
    }
```

hideakitai, 2023-07-31:
Thanks for your excellent knowledge!

closed 2023-07-31 by hideakitai